sccn/eeglab #775: an issue with clean_artifacts

arefpz, 2024-05-28:
### Description

Hi,

While I was using clean_artifact from clean_rawdata, I faced an issue. I may be doing it wrong but I'll explain it here.
If we have not-empty channels (which is a cell) and empty channels_ignore as of lines 207-217 I will have oriEEG, and the lines 218-226 won't be evaluated since channels_ignore is empty. So, I won't have "oriEEG_without_ignored_channels". But in lines 324-327, the first condition will be true (if ~isempty(channels) || ~isempty(channels_ignore)). if the second condition is true, it will ask for "oriEEG_without_ignored_channels" which is unavailable.
This generates an error.
<br>
#### Steps to Reproduce

1. pop_loadset(filename)
2. EEG=clean_artifacts(EEG, 'ChannelCriterion',0.85,'LineNoiseCriterion',4,	'Channels_ignore',	[],'Channels',{'F3','F4','C3','C4','P3','P4','F7','FP1','FP2','F8','FZ','CZ','PZ','VEOG','A2','HEOG'},'BurstCriterion',5,'WindowCriterion',0.25,'Highpass',[0.25,0.75],'ChannelCriterionMaxBadTime',0.5,'BurstCriterionRefMaxBadChns',0.075,'BurstCriterionRefTolerances',[-inf,5.5],'BurstRejection','off','WindowCriterionTolerances',[-Inf,7],'FlatlineCriterion',5,'NoLocsChannelCriterion',0.45,'NoLocsChannelCriterionExcluded',0.1);

<br>
#### Expected behavior:
I expect to get cleaned EEG data.

<br>
#### Actual behavior:
Unrecognized function or variable 'oriEEG_without_ignored_channels'.

Error in clean_artifacts (line 326)
        removed_channels = { oriEEG_without_ignored_channels.chanlocs(removed_channels).labels };
 
 
_----_ Please note I used both current version of clean_artifact and the one available on github. the line number in my current version was 317.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     |Windows11|
| Matlab version |R2023b |
| EEGLAB version | 2023.1 |

arnodelorme, 2024-06-18:
Can you try replacing the attached function and let us know if that fixes the problem?

https://github.com/sccn/clean_rawdata/blob/master/clean_artifacts.m

arnodelorme, 2024-07-02:
Let us know. Closing for now.

closed 2024-07-02 by arnodelorme

arefpz, 2024-07-15:
It worked and fixed the error. Thanks.

arnodelorme, 2024-07-15:
Thanks for letting us know.