sccn/eeglab #776: numerous bugs

LucieAman, 2024-05-30:
### Description
I am facing numerous matlab errors while using EEGlab for functions that, I think, are EEGlab functions.  I have used EEGlab to analyse about 40 datasets these past months without major problems. I have recently changed my laptop,  and the scripts are not working anymore, and I have errors I've never seen before. I suspect it is because of the new laptop (Macbook pro M3 chip, OS Sonoma14.2 ) but I don’t know what is wrong and how to fix it. I have tried to install newer versions of Matlab and EEGlab, but also old versions, and it still does work.
 
So, what I am trying to do is to import MFF files, then epoch and filter them, but I am stuck at both of these steps. I have tried to use the GUI to import data, but then when I try to filter and epoch using the script, matlab is still throwing an error, mostly saying that no epochs can be found (see screenshot attached).
I have attached the main script (pipelineMMN), and the functions used to import (dataimport_MMN), filter and epoch (epoch_MMN), as well as one of my data files (PiPWS_PWS20_GloLoc). The files can be downloaded from this link https://we.tl/t-PEaRzTNe8v 
[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*] change the path on the LoadPathMMN script and create Raw_data folder, Imported folder and Epoched folder
2. [*Second Step*] run the Setup directories and get filenames section
3. [*and so on...*] run the Import section
4. run the Epoch and filter section

<br>
#### Expected behavior:
[*What you expected to happen*]
step 3 data is imported mff files are imported in the Imported folder in a .set and .fdt files
step 4 data is filtered and epoched and saved in Epoched folder. The different stimuli flags are kept 

<br>
#### Actual behavior:
[*What actually happened*] error in import section and filter section as on the screenshot

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [14.2] | 
| Matlab version | [ 2019b 2023b and 2024a] | I tried with 2019b 2023b and 2024a
| EEGLAB version | [2021 and 2024]

arnodelorme, 2024-05-30:
I have looked at your MFF file, and it is fine. All EEGLAB functions should be backward compatible. Unfortunately, we do not have the resources to configure and run your scripts and assist in debugging it.

closed 2024-05-30 by arnodelorme

LucieAman, 2024-05-30:
Dear Arnaud,
Many thanks for your prompt reply. I will try to find someone in my department to help with debugging.

Best wishes
Lucie

Envoyé à partir de Outlook pour iOS<https://aka.ms/o0ukef>