sccn/eeglab #777: wrong ICA label after using pop_iclabel

thchabin, 2024-06-07:
Hi, 

I recently experienced wrong ICA labelling after applying decomposition with runica and using then the pop_iclabel function. The decomposition seems right based on topographies and signals but it does not give the right labels.

I did not change any parameters but after moving to Windows 11 and having reinstalled eeglab it was not working well anymore.

I run this ICA on a biosemi 64 channel dataset (average ref and filtered), I checked the channel loc file which is right. Here is an example of what I get for basic vertical and horizontal eye artifacts : 

[image]
[image]

If you have any idea of what happens, thanks for your help. 

Thibault.

arnodelorme, 2024-06-18:
This was fixed in ICLabel 1.6. Can you update the plugin from the command line?

thchabin, 2024-06-18:
Indeed I missed the update, the issue is fixed, thanks.

closed 2024-07-02 by arnodelorme