sccn/eeglab #778: ICLabel picks up right ear source as an eye component in eeglab2024.0

MakotoMiyakoshi, 2024-06-16:
See the comparison between 2022.1 (left) and 2024.0 (right.

[image]

KlausGramannTUB, 2024-06-17:
we observed the exact same thing and resolved the issue after reinstalling ICLabel. Let me know whether this also works for you.

arnodelorme, 2024-06-17:
This was fixed in the newer version of ICLabel. Can you update ICLabel?

Arno

MakotoMiyakoshi, 2024-06-17:
Arno, when I downloaded the latest version of EEGLAB, this broken ICLabel
came with it.

Makoto

arnodelorme, 2024-06-17:
Yes, I know, but then you have a red message to update ICLabel when you start EEGLAB.

Arno

MakotoMiyakoshi, 2024-06-17:
I disagree with your solution, but understand that the error has been
already taken care of.
I would agree with you if the red message would show "DANGER! YOU PUBLISH
THE ICLABEL RESULTS AND YOU WILL REGRET IT FOREVER" in 40 point font size.
It's because users can't know the severity of the problem the update took
care of.

Makoto

arnodelorme, 2024-06-17:
Yes, I could overwrite it in EEGLAB 2024.0 but then some versions of EEGLAB 2024.0 would be different from others. I plan to release 2024.1 early July and then we can have a warning message in red.

Arno

closed 2024-08-07 by arnodelorme