sccn/eeglab #781: a presicion issue of pop_writeeeg function when saving data to bdf format

hao199509, 2024-07-01:
### Description

[*when i use command "pop_writeeeg(EEG1, save_path_bdf, 'TYPE','BDF');" to save data to BDF format, i find overbit occur .*]

<br>
#### Steps to Reproduce

1. [*First Step:read an edf format eeg data *]
2. [*Second Step:save to BDF format*]

<br>
#### Expected behavior:
[*save the bdf data  same with the edf data without any loss*]

<br>
#### Actual behavior:
[*i read the bdf data saved before,it have some different with the edf data,when i observe the new bdf data,i find same continous point have the same value,i guesse that overbit happened in saving process ;
i watch the writeeeg function,find that there is a fix about bdf precision inline 211-215,but the code ,in 246-250 line,cover its function.
so when i save the bdf format,actually,saved as 16bit int liking edf format.
at last,i add the follow code after 250-line,it works as my expection.
"if strcmp(HDR.TYPE, 'BDF')
    HDR.DigMin  = repmat(-2^23, size(HDR.PhysMin));
    HDR.DigMax  = repmat(2^23-1, size(HDR.PhysMax));
end"
*]

<br>
#### Versions

|   |   |
[image]

|----| ---- | 
| OS version     | [*windows 11*] |
| Matlab version | [*2022a*] |
| EEGLAB version | [*eeglab2022.0*] |

arnodelorme, 2024-07-01:
@schloegl this is a potential problem with BIOSIG. Would you mind to answer?

schloegl, 2024-07-09:
Please send a patch against upstream and I'll consider adding it
 https://git.code.sf.net/p/biosig/code