sccn/eeglab #784: Array indices must be positive integers or logical values

GIlbertoGalindoAldana, 2024-07-08:
Hi! I am having trouble to PRECLUSTER my eeg files from OPENBCI.

My datasets do not have any events on them, I was able to create a STUDY, and previously followed all steps before preclustering. However, in the end, same error comes up, Array indices must be positive integers or logical values.

Before importing my raw data I've eliminated the rows that do not have EEG data (%channel, %Hz) to be able to import it, otherwise, EEGLAB does not read it. 

Here are a couple of my datasets used to create a STUDY, where I get the error: Array indices must be positive integers or logical values

[Areli_Villafuerte_BASELINE.txt](https://github.com/user-attachments/files/16134680/Areli_Villafuerte_BASELINE.txt)
[Ariana_Franco_BASELINE.txt](https://github.com/user-attachments/files/16134682/Ariana_Franco_BASELINE.txt)

GIlbertoGalindoAldana, 2024-07-08:
Hello again, 
Somehow I've solved it, but now the error is different. Unrecognized fieldname "comp6" 
When I try to PRECLUSTER for power spectra.

arnodelorme, 2024-07-11:
Can you share your STUDY?

SuprajaVaidhyanathan, 2024-11-14:
Hello, I am getting the same error. My data is resting baseline as well.

[image]

SuprajaVaidhyanathan, 2024-11-14:
> Hello again, Somehow I've solved it, but now the error is different. Unrecognized fieldname "comp6" When I try to PRECLUSTER for power spectra.

Mine is in 'comp49'

arnodelorme, 2024-11-14:
I would need access to the study.

SuprajaVaidhyanathan, 2024-11-16:
​Sorry, due to IRB, I cannot share patient data. Could you try to debug using
another open source resting dataset?  

::DISCLAIMER::

---------------------------------------------------------------------
The 
contents of this e-mail and any attachment(s) are confidential and
intended 
for the named recipient(s) only. Views or opinions, if any,
presented in 
this email are solely those of the author and may not
necessarily reflect 
the views or opinions of SSN Institutions (SSN) or its
affiliates. Any form 
of reproduction, dissemination, copying, disclosure,
modification, 
distribution and / or publication of this message without the
prior written 
consent of authorized representative of SSN is strictly
prohibited. If you 
have received this email in error please delete it and
notify the sender 
immediately.
---------------------------------------------------------------------
Header of this mail should have a valid DKIM signature for the domain 
ssn.edu.in <http://www.ssn.edu.in/>