sccn/eeglab #785: install biosig plugin

amisepa, 2024-07-10:
Just did a pull and tried to load a bdf file. When prompted to install biosig by eeglab (because it wasn't installed), I clicked Yes, and got this error.

[image]

amisepa, 2024-07-10:
and this:
[image]

arnodelorme, 2024-07-10:
Which version of EEGLAB are you using? The development version?

amisepa, 2024-07-10:
Yes, develop

amisepa, 2024-07-10:
I also get this warning when I launch eeglab, even though I just did a pull on develop. Normal?

[image]

arnodelorme, 2024-07-10:
I just removed biosig, restarted EEGLAB, then try to open a BDF, installed BIOSIG, and imported the BDF. 

Yes, warning message is OK. I am releasing version 2024.1 but it is not registered yet.

arnodelorme, 2024-07-10:
Can you try again
- Do a pull
- Quit MATLAB
- Remove BIOSIG folder
- Start MATLAB
- Import BDF

amisepa, 2024-07-10:
same error. I even reinstalled eeglab because I have some changes I couldn't discard.

amisepa, 2024-07-10:
in case it's helpful, I was able to install bdf_import plugin, so the error is specific to biosig

arnodelorme, 2024-07-10:
We would need to get on a quick call to look into that. Are you available now?

amisepa, 2024-07-10:
solved

closed 2024-07-10 by amisepa