sccn/eeglab #786: Bon jour j'ai un problème c'set que l'icone ERP n'est pas active comment faire ?    Merci

Turkia7, 2024-07-10:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

Turkia7, 2024-07-10:
Bon jour j'ai un problème c'est que l'icone ERP n'est pas active comment faire ? Merci

arnodelorme, 2024-07-10:
You might need to install ERPLAB? https://erpinfo.org/erplab