sccn/eeglab #789: rereference the data after ica + interpolation error message

lulyluly, 2024-07-19:
I do not include noisy channels in ICA which I only do to remove eye related components, and specifically want to interpolate then after ICA for rank reasons. I have now done this to over 100 datasets.
However, once I interpolate and then re-reference my data to the average, I get the error: Error: some channels not used for ICA decomposition are used for rereferencing
       the ICA decomposition has been removed

However, I purposefully wanted to interpolate these after ICA, and then use them for the average.
Is my eye-blink removal via ICA now affected and undone?
I would appreciate any advice.
Best wishes,

arnodelorme, 2024-07-19:
To do what you want to do, you would need to interpolate the scalp topography of each component onto the additional electrodes. Then you will be able to reference. 

Once you interpolate ICA component scalp topographies, it is no longer ICA.

Another approach is to interpolate the channels before ICA, and then run ICA with a reduce rank, the re-reference after ICA.

closed 2024-07-19 by arnodelorme

lulyluly, 2024-07-19:
Thank you very much for your reply 🙏

I tried another approach: to reference )compute average) .- run ICA (
excluding removed  electrodes ) and then  interpolation.
It’s ok?

(Removed/ bad channels in the most of cases because of cochlear implants
processor or coils)