sccn/eeglab #791: using ADJUST plugin

lulyluly, 2024-07-24:
hi 
How can i use the Adjust plugin after applying ICA if I excluded some channels (surrounding cochlear implant processor) 
i received a message stating that I had not run ICA on all channels, even though I attempted to interpolate them before running ICA.

thank you

vpKumaravel, 2024-08-03:
@marcobuiatti, maybe you could help here?

marcobuiatti, 2024-08-05:
Hi Luly,

I am not sure I fully understood your problem. If you removed some channels, you should not interpolate them before running ICA but after you ran ICA and removed artefacted IC components with ADJUST. Does this answer your problem?

Best,

Marco

Thanks @vpKumaravel

346836814, 2024-09-22:
@marcobuiatti 
hello,Marco.
I want to know why bad  channels interpolation should be carried out after ica runs. Can you tell me? Are there any related articles or documents?
best.
kaixin_huang.

marcobuiatti, 2024-09-23:
Dear Kaixin,

If you interpolate before ICA, you need to tell ICA that the rank of your data is smaller than the number of channels. Also, I seem to remember some messages in the EEGLAB list saying that it is better to not interpolate before for a better ICA decomposition (but I cannot find where). If you dig in the EEGLAB mailing list you should find such references.

Best,

Marco

arnodelorme, 2025-02-07:
Closing for now.

closed 2025-02-07 by arnodelorme