sccn/eeglab #793: Error in plugin_menu at line (52)

antB16, 2024-07-25:
[image]
### Description

[*Error message "EEGLAB error in funtion plugin_menu() at line 52:*]
Unrecognized field name "downloads"
<br>
#### Steps to Reproduce

1. [*eeglab*]
2. [*File---Manage EEGLAB extensions*]

<br>
#### Expected behavior:
[*Open extension options and download extension that I require i.e MICROSTATELab*]

<br>
#### Actual behavior:
[*Cannot view plugins at all*]

<br>
#### Versions

| OS version     | [*Windows 10*] |
| Matlab version | [*2022b*] |
| EEGLAB version | [*2024*] | latest version was updated today. 
I have not had this issue with the previous version that I was using (2022).

Steps taken so far:
1. Uninstalled and re-installed EEGLAB from the sccn website
2. Used a different internet connection
3. Used a different version of MATLAB 2022a and 2022b

antB16, 2024-07-26:
Update on the issue: Inspected the plugin_menu function, and ran it separatley.
line (42) : plugin = plugin_getweb('plugin_check', pluginlist); returns an error that says : Plugin list incomplete - maybe an issue with internet connection - aborting. 
(Internet is not an issue, I have checked with 3 different connections in two locations)
pluginlist (in caps) contains all the plugins in the main folder: 
[image]
[image]

arnodelorme, 2025-02-07:
I have implemented a special test in 5452f98fa

closed 2025-02-07 by arnodelorme