sccn/eeglab #794: spam email associated to ADJUST plugin!

marcobuiatti, 2024-08-08:
### Description

The email associated to ADJUST in the EEGLAB plugin list https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php
is false, likely a spam! The correct email is mine: marco.buiatti@unitn.it as I am the inventor and manager of ADJUST. Also, the description of ADJUST is vague and should be updated.

I therefore kindly ask you to modify the reference to ADJUST in https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php in the following way:
- replace the contact of ADJUST with mine (Marco Buiatti, email: marco.buiatti@unitn.it)
- Name ADJUST in all capital letters as it is an acronym (ADJUST, not Adjust)
- Replace the description of ADJUST with this one: "Automatic identification and removal of EEG artifacts based on ICA"

Thank you!

Best,

Marco  

<br>

arnodelorme, 2024-08-08:
Hi Marco

> replace the contact of ADJUST with mine (Marco Buiatti, email: [marco.buiatti@unitn.it](mailto:marco.buiatti@unitn.it))

Done

> Name ADJUST in all capital letters as it is an acronym (ADJUST, not Adjust)

This is possible, but you would lose all the download stats. We would also need to redo the plugin submission https://sccn.ucsd.edu/eeglab/plugin_uploader/upload_form.php

> Replace the description of ADJUST with this one: "Automatic identification and removal of EEG artifacts based on ICA"

Done

marcobuiatti, 2024-08-09:
Hi Arno,

> Hi Marco
>
> replace the contact of ADJUST with mine (Marco Buiatti, email:
>
> Done
Thanks!

> Name ADJUST in all capital letters as it is an acronym (ADJUST, not Adjust)
>
> This is possible, but you would lose all the download stats. We would also
> need to redo the plugin submission
> https://sccn.ucsd.edu/eeglab/plugin_uploader/upload_form.php
Ok, this is not necessary then.

> Replace the description of ADJUST with this one: "Automatic identification
> and removal of EEG artifacts based on ICA"
>
> Done
Thanks!

Best

Marco

closed 2025-02-07 by arnodelorme