sccn/eeglab #796: FileIO plugin dot indexing

serotoninkid, 2024-08-14:
### Description

Issue when installing the 'Fileio' plugin for EEGLAB. Plugin installs just fine, but displays issue when trying to add Fileio plugin to path. 

[image]

<br>
#### Steps to Reproduce

1. install EEGLab v2024.1
2. File -> import data -> import using FILEIO interface -> prompted to install FILEIO plugin
3. plugin is installed -> launch EEGLAB -> issue appears

<br>
#### Expected behavior: Plugin should be installed, added to path and then be used to import .raw files
[*What you expected to happen*]

<br>
#### Actual behavior:
Plugin was installed and whilst EEGLab adds plugin to path, the issue comes up and EEGLab initialisation cannot proceed any further

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 11 Home. Version - 23H2. OS Build - 22631.3880   |
| Matlab version | 2023a |
| EEGLAB version | 2024.1 |

serotoninkid, 2024-08-14:
Hi! 

Internet was definitely working at the time of the error as going into eeglab -> plugins I can see the Fileio folder complete with the files it needs to work. 

I downgraded a few versions of EEGLab to version 2022 and I can't seem to replicate the error again.

arnodelorme, 2024-08-14:
But can you still replicate with 2024.1?

christophezxc, 2025-01-08:
I met this exact problem too.
System: Windows 10 Pro 22H2
EEGLAB version: 2024.2
Matlab version: 2023a

arnodelorme, 2025-01-08:
I believe this was fixed in EEGLAB version 2024.2.1

closed 2025-02-07 by arnodelorme