sccn/eeglab #797: MFF files imports not working

LucieAman, 2024-08-15:
### Description
I cannot import my mff files anymore. I have tried different version of Matlab, EEGlab, mffimport plugin. I have also increased the JAVA memory allowance. 
https://drive.google.com/drive/folders/19tWVFEuutKhhr35dd4ZH11kB5a_NSHjF?usp=drive_link 

<br>
#### Steps to Reproduce

1.  put the PiPWS_PWS023 mff file in a Raw_data folder
2.  change the loadpathMMN.m script with your path
3. run the first two sections of the pipelineMMN.m script

<br>
#### Expected behavior:
the mff file is imported in a .set and .fdt files in the Imported folder

<br>
#### Actual behavior:
[*What actually happened*]
Error using mff_importsignal (line 218)
Error: Can not open the signal resource.\n

Error in mff_import (line 95)
[floatData, allDataSize, srate, nchans] = mff_importsignal(mffFile);

Error in pop_mffimport (line 119)
    EEGTMP = mff_import(fileName{iFile}, correctEvents);

Error in dataimport_MMN (line 16)
EEG = pop_mffimport(sprintf('%s%s', filepath, filename), 'code'); %needs
MFFMatlabIO2.01 plugin

Error in pipelineMMN (line 49)
    dataimport_MMN(basenames{i}, raw_dir)

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Sonoma 14.2] |
| Matlab version | [*2024a and I have also tried with 2019b] |
| EEGLAB version | [*2024 and 2021*] |

CPernet, 2024-08-15:
just to be sure you have the latest -- can you remove your plugin and clone instead https://github.com/arnodelorme/mffmatlabio -- this one has lately worked well for me

LucieAman, 2024-08-15:
Dear Cyril,

Thank you for your help.
I have downloaded the pluging from the link you sent and it still doesn’t work. I get the same error message. I tired to put it in my MATLAB toolboxes folder and in the plugin folder of EEGlab2024. Is there any other place I should try?

Best wishes,
Lucie

CPernet, 2024-08-15:
alright, so I can see from the error something like

- your script dataimport_MMN runs the line 16 EEG = pop_mffimport(sprintf('%s%s', filepath, filename), 'code'); 
so you need to know (1) if indeed it calls the right file name in sprintf('%s%s', filepath, filename)

next, the error in pop_mffimport (line 119) EEGTMP = mff_import(fileName{iFile}, correctEvents);
could be that it does not find 'code' -- ah we pushed a fix recently, under review, see [the file here --> ](https://github.com/arnodelorme/mffmatlabio/pulls) and maybe pulling my student repository will just do the trick (because we also ran into a few subject with 'code' but read as event.code for some reason)

LucieAman, 2024-08-15:
Dear Cyril,

Thank you for your reply. I have checked that the right filename is called.
I have copied and pasted the code displayed at the link you sent me and replaced the pop_mffimport.m script with it in the mff toolbox. Was it the right thing to do?
I still have the same error though.

Best wishes,
Lucie

CPernet, 2024-08-16:
yes that was the right file but clearly did not help ... then you have to debug pop_mffimport.m  
`EEGTMP = mff_import(fileName{iFile}, correctEvents);`
my guess is correctEvents is faulty, then @arnodelorme should be able to find the fault
TO DO: 
- [ ] report on your discovery about `correctEvents`
- [ ] if possible, put a file on Google Drive to share

LucieAman, 2024-08-16:
Thank you Cyril for your help.
Please find below the result of correctEvents 
>> EEGTMP = mff_import(fileName{iFile}, correctEvents);
Undefined variable 'fileName'.
 
Did you mean:
>> EEGTMP = mff_import(mfilename{isfile}, correctEvents);
Brace indexing into the result of a function call is not supported. Assign the
result of 'mfilename' to a variable first, then brace index into it.
 
You can download a file here: https://drive.google.com/drive/folders/19tWVFEuutKhhr35dd4ZH11kB5a_NSHjF?usp=drive_link

LucieAman, 2024-08-26:
@CPernet, is there anything I can do to try to solve my problem? It is becoming increasingly urgent for me to import and preprocess my files. Is there any other alternative to EEGlab to import mff files?
many thanks for your help

arnodelorme, 2024-08-26:
Can you share a file that cannot be imported. It is on Windows or Mac?

LucieAman, 2024-08-26:
It is Mac. None of my MFF files can be imported. Please find o ne of them here: https://we.tl/t-1yTQcEMFS0
Thank you.

arnodelorme, 2024-08-27:
I imported this file with no problem on my MAC.
Can you try importing the file manually (with the EEGLAB menu). 

Would it be possible that you are importing the wrong file name in your script?

LucieAman, 2024-08-28:
Dear Arnaud,

Thanks for trying. Would you mind sending me the imported file if you still have it so I can work on it?
I have tried to import manually using the GUI but when I do that I cannot epoch the file. I get an error saying ‘ no epoch found’.

I don’t think I am importing the wrong file name in my script. I have also tried with files previously successfully imported in the past and I still get the same error. Also, I have checked I have the right ‘basename’ when I try to import it.
Would it be too much to ask for you to import a few more files on your laptop for me?  I only need to import 3 more files to be able to finish my analysis. No worries if not.

Best wishes,
Lucie

LucieAman, 2024-08-30:
Dear Arnaud and Cyrille,

I was wondering if you know a student I could pay to import  my 4 mff files for me? I urgently need to finish my eeg analysis, and I have no other options. Please let me know if you know anyone who would be interested and how much they would ask to do that. I have French and UK bank accounts but I can send a Wish bank transfer in any currency.

Best wishes,
Lucie

LucieAman, 2024-09-03:
Dear @CPernet and @arnodelorme I was wondering if you had seen my last message. Apologies for bothering you, but I have exhausted all my options. 
I have tried to import my files manually or with a different script and it worked, but I then have another error when I try to epoch my files: Error using pop_epoch (line 279)
pop_epoch(): empty epoch range (no epochs were found).
I know that there are epochs in my files.

Many thanks for what you can do to help.
Best wishes,
Lucie

arnodelorme, 2024-09-09:
Dear @LucieAman , if we cannot reproduce the issue, we cannot help. I would advise to make sure you have the latest version of EEGLAB and of the mffplugin.

closed 2024-09-09 by arnodelorme