sccn/eeglab #799: Edit Event Values

OswArti, 2024-08-20:
### Description

I am using EEGMATLAB on my MacBook, but the Edit Event Values menu item does not work; the pop-up window does not show. 

<br>
#### Expected behavior:
The pop-up window to show.

<br>
#### Actual behavior:
 the pop-up window does not show

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Sonoma 14.2.1] |
| Matlab version | [R2024a] |
| EEGLAB version | [v2024.1] |

carrie372, 2024-08-27:
Same issue using Windows 11

carrie372, 2024-08-27:
@arnodelorme I'm using EEGLAB the MATLAB version. I cannot share the dataset however when I use the 2024.0 EEGLAB version on the same dataset the edit event values menu item works properly. Using different datasets the edit event values pop-up window does not work in v2024.1.

OswArti, 2024-08-28:
> You mean EEGLAB, right? Are you using the compile or MATLAB version? Can you share the dataset you are using the exact steps to reproduce the problem?

Yes, EEGLAB with MATALAB. The problem is while I was doing the examples of the tutorial, for example, Preprocess data

| 
| 
|  | 
Preprocess data

EEGLAB Documentation including tutorials and workshops information
 |

 |

 |

 Oswaldo Artiles

arnodelorme, 2025-02-07:
Closing for now as we do not have enough info to look into the problem.

closed 2025-02-07 by arnodelorme