sccn/eeglab #802: Issue in plotting ERSP and ITC

jeenalrambhia, 2024-08-28:
I am running a study in EEG lab. I wish to plot ERSP and ITC plots but i am not able to do so in times vs frequency plots. On the other hand  i am  able to plot them in topography plot. Kindly let me know what i am doing wrong some message receive is 
Warning: std_readdata ignored at least one parameters: "subbaseline"
[image]

zaeemhadi, 2024-09-01:
Just bumping this as I have had a similar issue in the latest EEGLAB version as well and cannot use 2024.1.
Reported previously as below:
https://github.com/sccn/eeglab/issues/795

arnodelorme, 2024-09-01:
This is because you are using MATLAB earlier than 2020. Can you edit the icadef.m function and change the default colormap to "jet" line 70.

arnodelorme, 2025-02-07:
Can you share your study. Sorry for the late response.