sccn/eeglab #807: EEGLAB error in function pop_biosig() at line 139;pop_biosig;Unsuported data format

JXnju, 2024-09-12:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2025-01-23:
We would need more information. Apologies for the late answer.

Metanome, 2025-06-29:
I am experiencing the same issue when trying to import any EDF data file from this specific EEG database (https://github.com/OpenNeuroDatasets/ds005385) using the BIOSIG toolbox.

arnodelorme, 2025-06-29:
Any dataset in particular within [ds005385](https://github.com/OpenNeuroDatasets/ds005385)?

Metanome, 2025-07-01:
> Any dataset in particular within [ds005385](https://github.com/OpenNeuroDatasets/ds005385)?

I tried to import both the pre- and post-EyesClosed EDF files from these two subjects: https://github.com/OpenNeuroDatasets/ds005385/tree/main/sub-001/ses-1/eeg, https://github.com/OpenNeuroDatasets/ds005385/tree/main/sub-001/ses-2/eeg.

I also tried a random subject (around 400s), but I encountered the same issue.