sccn/eeglab #810: Discussion with Yahya about merging multiple steams

arnodelorme, 2024-09-16:
- See https://github.com/xdf-modules/xdf-EEGLAB

Have a function to merge multiple streams in eeg_load_xdf and use the eeg_mergechannels function on the clock times (in streams{1}.info.clock_offsets.offset). For each stream add the time and offset (in seconds) and compute the linear regression between streams (create an event table using these clock times and use eeg_mergechannels to align the streams).

@neuromechanist

neuromechanist, 2024-09-16:
Current plan:
1. Make a dropdown menu in the import GUI for the stream available in the XDF.
2. Synchronize the streams with an option for remapping to a base stream.
3. Put the resampled streams under the etc field.

arnodelorme, 2024-10-31:
Any progress @neuromechanist

neuromechanist, 2024-11-01:
Not much currently, I will circle back to this next Monday.

arnodelorme, 2025-01-23:
Done in the xdf-EEGLAB plugin.

closed 2025-01-23 by arnodelorme