sccn/eeglab #811: EEGLAB error in function pop_select_comps_AD() at line 168.The error message is that he number of input parameters was insufficien。

346836814, 2024-09-21:
my version of eeglab is 13.6.5b and adjust is 1.1.1. When I ran the ADJUST plugin, I encountered an error. How can I resolve this issue?
the figure shows the error.
When I ran the ADJUST plugin, it displayed an error: 'EEGLAB error in function pop_select_comps_AD() at line 168.' The error message is that he number of input parameters was insufficien。
[image]

there is a config figure.
 
[image]

firstly，i clicked the "ok" button.
[image]
then,the error was coming;
[image]

346836814, 2024-09-21:
I obtained the report.txt file, but after comparing its results with those from the IClable plugin, I believe the results after running ADJUST are problematic.

arnodelorme, 2024-09-25:
Yes, these are two different plugins. I have not used ADJUST so I do not know how to answer. Perhaps you should contact ADJUST authors or try a more recent version of EEGLAB.

closed 2024-09-25 by arnodelorme

346836814, 2024-09-25:
Ok，I understand it.Thank you.