sccn/eeglab #814: The requested array exceeds the maximum possible variable size

PhD-GOAT, 2024-09-30:
### Description

Memory error while importing EEG file?

<br>
#### Steps to Reproduce

1. impot data from Neuroscan .EEG file
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
Expect .EEG files to be imported.

<br>
#### Actual behavior:
[image]
An error occurred while using: zeros
The requested array exceeds the maximum possible variable size.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [window 11] |
| Matlab version | [R2024b] |
| EEGLAB version | [v2024.2] |

arnodelorme, 2024-09-30:
Are you sure it is a neuroscan file. Different manufacturer use the .eeg extension.

PhD-GOAT, 2024-09-30:
> Are you sure it is a neuroscan file? Different manufacturers use the .eeg extension.

Thank you for telling me what I missed. It is a 128-channel ActiveTwo EEG system (BioSemi), and it is an .EEG file. I installed bdfimport and tried to import it, but a format error appears.

arnodelorme, 2024-10-31:
You might want to rename your file to BDF. ActiveTwo biosemi files are BDF files. Closing for now, but feel free to reopen.

closed 2024-10-31 by arnodelorme