sccn/eeglab #815: error in eeglab.m if there are no "pluginstats"

olafdimigen, 2024-10-15:
### Description

Tried to install Biosig extension (v3.8.4.) under EEGLAB 2024.2. Starting EEGLAB afterwards consistenly produces the following error:

```
EEGLAB: adding "Biosig" v3.8.4 to the pathDot indexing is not supported for variables of this type.
Error in eeglab (line 1111)
                indPlugin = strmatch(lower(pluginlist(plugincount).plugin), lower({ pluginstats.name }), 'exact'); 
```
Note that earlier in `eeglab.m`, there is code to explicitly set `pluginstats` to empty, which seems to happen with the Biosig extension, at least when it is installed not via the Extension manager (maybe always):
```
        if ~isfield(pluginstats, 'name') || ~isfield(pluginstats, 'version')
            pluginstats = [];  % you are setting it to empty here
        end
```
However setting `pluginstats = []` looks will invariably cause a crash in `eeglab.m` later on, when you assume the presence of a field `name `(about line 1111):

```
indPlugin = strmatch(lower(pluginlist(plugincount).plugin), lower({ pluginstats.name }), 'exact');
% there is no subfield name in an empty pluginstats, thus causing an error!
```

|   |   |
|----| ---- | 
| OS version     |  Windows 10|
| Matlab version |  2023b |
| EEGLAB version | 2024.2 |

arnodelorme, 2024-10-31:
Hi Olaf, apologies for the late answer and thank you for reporting. This was fixed in 2024.2 (but not 2024.1). Can you double check you are using 2024.2. At line 1111, I have

```matlab
                if ~isempty(pluginstats)
                    indPlugin = strmatch(lower(pluginlist(plugincount).plugin), lower({ pluginstats.name }), 'exact');
                else
                    indPlugin = [];
                end
```

olafdimigen, 2024-11-01:
Hi Arno,
I checked, the version was 2024.2, not 2024.1:

```
vers        = '2024.2';
releaseDate = '28-Aug-24 14:55:42'; % unix date -> date +"%d-%b-%y %T"
```

arnodelorme, 2024-11-01:
OK, thanks Olav, let me redownload 2024.2 and check myself.

arnodelorme, 2024-11-04:
I was able to reproduce the problem. The issue arises when using EEGLAB offline. The issue is fixed in the develop branch. Because it is a big enough problem, I will issue version 2024.3, but first, I need to make sure it passes all of the tests, so it might take a few days. Thanks Olav

olafdimigen, 2024-11-05:
Yes, I likely used it offline due to bad WiFi on trains. 
I hotfixed it for myself, thanks for addressing it in a future release.

arnodelorme, 2024-11-05:
Thanks Olaf for reporting.

closed 2024-11-05 by arnodelorme