sccn/eeglab #821: pop_chanedit fails to append reference channel

carvidarb, 2024-10-29:
### Description

I downloaded a matlab toolbox to analyse eeg data, additionally to eeglab. This toolbox is used to analyse eeg data from people who experience pain. This toolbox is based on EEG lab and I am using it with their demo data. In this code pop_chanedit fails to add the reference channel, in the EEGtemp structure that is returned, chanlocs.labels are all 'E', but in the original EEGtemp, the labels in chanlocs are correct. I could track the problem until eeg_checkset in line 1081. At that point, the field labels is correct and a field called label has appeared and is empty. Then inside eeg_checkset, line 272, the field labels is interchanged with the field label, which is empty and messes up the labels field. 
The channel to insert is A1 (the reference). In the main script of the analysis toolbox it is written:

<br>
#### Steps to Reproduce

1.  EEGtemp = eeg_checkset(ALLEEG(iRec),'loaddata');
2.  EEGtemp = pop_chanedit(EEGtemp, 'append',EEGtemp.nbchan, ...
            'changefield', {EEGtemp.nbchan+1,'labels',ALLEEG(iRec).BIDS.tInfo.EEGReference},...
            'changefield', {EEGtemp.nbchan+1, 'X', params.RefCoord.X}, ...
            'changefield', {EEGtemp.nbchan+1, 'Y', params.RefCoord.Y}, ...
            'changefield', {EEGtemp.nbchan+1, 'Z', params.RefCoord.Z},...
            'setref',{['1:' num2str(EEGtemp.nbchan)],ALLEEG(iRec).BIDS.tInfo.EEGReference});

without knowing very well what I am doing (I am new to eeglab), I added in the main script:
for ichan=1:length(EEGtemp.chanlocs)
         EEGtemp.chanlocs(ichan).label=EEGtemp.chanlocs(ichan).labels;
        end
between steps 1 and 2. In this case, the updated EEGtemp after pop_chanedit, has the correct chanlocs.labels, but the reference channel is not added to the rest.

<br>
#### Expected behavior:
I expect that the reference channels is added without changing any field of the other channels, apart from setref (I suppose)

<br>
#### Actual behavior:
The labels in chanlocs are messed up, all are the same and it is 'E'. If I write the additional code, the labels are fine (because eeg_checkset cannot mess them up, but the reference channel is not added.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*14.6.1*] |
| Matlab version | [*2023b*] |
| EEGLAB version | [*2024..2*] |

arnodelorme, 2024-10-31:
Thanks for reporting. EEG.chanlocs(x).label is not supposed to be a field in EEG.chanlocs. I am not sure why this section of the code of eeg_checkset is executed on your side. It is meant for future versions of EEGLAB where EEG.chanlocs.labels will be renamed EEG.chanlocs.label (without s), because the s should have never been there in the first place. However, for the current version, it should have no incidence.

carvidarb, 2024-11-04:
Thank you for your answer... what do you suggest me? remove eeg_checkset in pop_chanedit? Or this is not possible? Sorry, I am new to eeglab and I do not know how to proceed properly. Alternatively, I can set the reference channel by "hand" instead of using pop_chanedit, if you think this will not cause a problem... Sorry for insisting, I am a bit clueless

AllyGwl, 2024-11-12:
Hello. I have the same issue. Whenever I try to add a channel (using the Append or Insert button in Edit/Channel Locations, or using pop_chanedit in a script) I always end up with all the labels changing to 'E'.
Is there a workaround for this, please? Thanks.

arnodelorme, 2024-11-12:
Yes, EEGLAB will rename your channels to E1, E2, etc. if your labels are 1, 2, 3, etc. Is that what you mean?

AllyGwl, 2024-11-12:
No. Sorry. I meant that labels such as 'Fp1', 'Fpz', Fp2' etc. were all changed to 'E', 'E', 'E' after using pop_chanedit (via script or gui). I think I've fixed it in my local version by changing Line 686 in pop_chanedit from:
chans(end+1).label = '';
chans(end+1).labels = '';

arnodelorme, 2024-11-12:
The reference channels is set using the 'ref' field, not the 'labels' field

AllyGwl, 2024-11-12:
Any time pop_chanedit was called for any reason, it would fill EEG.chanlocs.labels with 'E' on every row. This didn't happen in previous versions of EEGLAB. I tried changing chans.label to chans.labels in pop_chanedit following your comment to carvidarb above. That seemed to fix it. I think it might just be a typo on Line 686 of pop_chanedit because the chans.label column doesn't exist before that line and isn't used anywhere else in the function.

arnodelorme, 2024-11-12:
Sounds good. Happy it is fixed. Yes, the correct spelling is "labels" not "label" (I know it is not logical).

AllyGwl, 2024-11-12:
Thanks. I've just sent a pull request. https://github.com/sccn/eeglab/pull/825
Hope that's OK.