sccn/eeglab #826: OpenBCI x Emotibit xdf file import error

Julluke, 2024-11-13:
Hi EEGLAB Community!

I have a measurement system consisting of Emotibit and the OpenBCI Ultracortex Mark IV. I have saved all data streams from both devices via LSL Stream in an xdf file. Now I have problems with the EEGLAB plugin “Mobilab”. When I try to import the xdf file, the widget opens, but it is empty (file attached). 
I also cannot run the "Load MoBi folder" command.
I'm using Matlab 2024 and the latest verison of eeglab and mobilab plugin, since i've read some posts about errors using non compatible versions, but didnt quite understand what i should change. Has anyone here tried a similar setup or had similar problems with the plugin?
Since i'm not dependent on using eeglab for this purpose, is there any other way of using Emotibit data with other data in an xdf file? 
 
Any suggestion would be helpful.

Thank you very much in advance :)

Jule

[image]

arnodelorme, 2024-11-13:
Yes this is a known issue. Can you try the EEGLAB xdf plugin?

Julluke, 2024-11-13:
Thanks for the quick reply!

I can import the data via "Import File -> xdf" of the EEGLAB GUI. But then only the 16 EEG channels can be seen. 
I also checked the xdf file in Matlab with "load_xdf". There can all data streams be plotted, but the 16 Channels of the eeg are not separate. And especially this would be the major benefit of EEGLAB, as i suggest. 

Thank you very much in advance for any help :)

Jule

arnodelorme, 2024-11-13:
We will try to remedy that problem, do you have a file to share?

Arno

Julluke, 2024-11-13:
Yes of course, here is the file i'm using. Please consider, this was only for testing the setup and no significant data is expected. 
This zip file should work: 
[sub-P001_ses-S001_task-Default_run-001_eeg.zip](https://github.com/user-attachments/files/17735729/sub-P001_ses-S001_task-Default_run-001_eeg.zip)

Thanks a lot for your effort :)

arnodelorme, 2024-11-13:
OK, thanks. We should have something next week.

Julluke, 2024-11-25:
Hi Arno, i kindly wanted to ask whether the issue has been fixed or any other solution has been found. 
I appreciate your work and any hint or solution will be helpful :)

Jule

arnodelorme, 2024-11-25:
Yes, please try the latest version of the plugin
https://github.com/xdf-modules/xdf-EEGLAB

Julluke, 2024-11-25:
Hi Arno, thank you very much but somehow it doesn't work like intended. 
I reinstalled the plugin via the eeglab plugin manager, but it is still the version 1.19. It is also the same when downloading the version of the Internet plugin list. The gui remains also the old one and not the one you show here on Github.
I also tried to manually update the codes provided here on Github, but then any xdf data cannot be found anymore. 
My version of eeglab is v2024.2 and the version of matlab is 2024.a (since there is a newer available since november). 

I'm very sorry to bother you again...

Jule

Julluke, 2024-11-25:
Hi Arno, thank you very much, i could import the streams. 
But only after i manually updated the codes provided her on Github. When reinstalling, it remains the old version of the plugin. 
I also have to change the folder in Matlab, which i didn't have to before. 
I can now plot the eeg Data parallel to another stream of my emotibit device, but only one of the 16 additional. if a try to add more, then an error occurs. 
Additionally, the streams beside the eeg data are incorrect (stay static or aren't visible). Maybe i have to plot all my data separated from each other...

I'm very sorry to bother you again...

Jule

arnodelorme, 2024-11-25:
[image]

I was able to import your data (all streams). You need to use the version on Github instead of the version on the plugin manager. The plugin will be released within 2 weeks on the plugin manager once we perform all the tests.

[image]

Your data is unusual. Is it OK to add the data to my testing set (it will be shared with other developpers).

Julluke, 2024-11-25:
Ok, thanks again. I will give it another try and get back to you if necessary.
Yes of course you can use the data 👍

arnodelorme, 2025-01-23:
Closing for now.

closed 2025-01-23 by arnodelorme