sccn/eeglab #829: Incorrect handling of OK button tag in inputgui.m

UchidaHitoshi, 2024-11-21:
### Description

There is a problem with the tag property of the OK button in the processing described in the functions/guifunc/inputgui.m file.

Because the following two different methods are mixed, it may not be possible to detect that the OK button has been pressed. Specifically, the Cleanline plugin does not work with EEGLAB 2024.2.

  1) Using the value of the g.okbut variable as the value of the tag property
  2) The fixed string "ok" is used as the value of the tag property.

In addition, inputgui.m sets the default value of g.okbut to "OK" in the upper case, which can cause problems if you use the value of the g.okbut variable as the value of the tag property.

The Text of the Push Button Object is rewritten in supergui.m with only the first character being uppercase and the second and subsequent characters being lowercase, so the value that was "O**K**" before the Figure was displayed is "O**k**" when the Figure is displayed. Therefore, if you press the OK button and then try to use findobj to retrieve data whose tag property is "O**K**", you will fail.

<br>

|   |   |
|----| ---- | 
| EEGLAB version | [2024.2] |

arnodelorme, 2025-01-23:
Correct. Thanks for reporting. We are working on the issue.

arnodelorme, 2025-09-10:
I have checked that Cleanline works. This was probably fixed a while ago.