sccn/eeglab #832: pop_adjustevents and boundary events

marcobuiatti, 2024-11-30:
Dear EEGLAB masters,

I am signalling a potential risk of the use of pop_adjustevents. 
- I used pop_adjustevents to correct a stimulus-trigger delay that I found out after pre-processing my EEG data;
- I applied it to continuous, preprocessed data containing boundary events corresponding to rejected bad data segments;
- In some cases, pop_adjustevents moved the stimulus event across the boundary event, erroneously marking an event on the subsequent "good" data segment.

Therefore, an option in pop_adjustevents to avoid this hidden risk could be very useful.

Hope this helps,

Best,

Marco

arnodelorme, 2024-11-30:
Thanks Marco, this will be addressed.

arnodelorme, 2025-02-07:
This has been addressed in d3e6b10b0

closed 2025-02-07 by arnodelorme