sccn/eeglab #833: the "scroll dataset" button in pop_select() seems to work only with the variable "EEG", not with, e.g., "myEEG" in pop_select(myEEG)

cmaidhof, 2024-12-02:
### Description

When I call pop_select from a script and press the "scroll dataset" button, it seems to always use variable EEG as input - even if I call pop_select(myEEGstructWithAnotherName). If there is a variable EEG in the workspace, it seems it will plot the data from that EEG variable, and not from the one given as a parameter with the function call. If there is no variable named "EEG" in the workspace, I get an error (Unable to resolve the name 'EEG.data'.)

<br>
#### Steps to Reproduce

1. load any dataset, for example data in eeglab sample data folder: myEEG = pop_loadset();]
2. if exist('EEG'); clear EEG; end 
3. [pop_select(myEEG)]
4. [press "Scroll dataset" button in dialog]

<br>
#### Expected behavior:
Maybe it is the correct behavior, although I would have expected that the input dataset (parameter from function call) would have been plotted, independently from the name. 

<br>
#### Actual behavior:
[If no "EEG" variable exists, an error is thrown. If "EEG" variable exists, this variable "EEG" will be plotted (instead of the variable given as the parameter).]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*WIN10 home*] |
| Matlab version | [*R2024a*] |
| EEGLAB version | [*2024.2.1*] |

Best, clemens

arnodelorme, 2025-01-07:
OK, we will look into this. Thanks for reporting

arnodelorme, 2025-02-07:
Fixed in 078452753

closed 2025-02-07 by arnodelorme