sccn/eeglab #834: Channel Locs Import Issue - " Brace indexing "

saptrshighosh, 2024-12-16:
### Trying to Import Channel Location File in EEGLAB

**Command Window** 
###################

Reading data file header...
sopen mode is "OVERFLOWDETECTION:OFF"
Reading data in EDF format...
eeg_checkset note: upper time limit (xmax) adjusted so (xmax-xmin)*srate+1 = number of frames
Detected/removing 'EEG' prefix from channel labels
Extracting events from last EEG channel...
eeg_checkset warning: number of columns in data (24) does not match the number of channels (25): corrected
eeg_checkset note: creating the original event table (EEG.urevent)
Creating a new ALLEEG dataset 1
Done.
IMPORTANT: After importing/modifying data channels, you must close
the channel editing window for the changes to take effect in EEGLAB.
TIP: Call this function directly from the prompt, ">> pop_chanedit([]);"
     to convert between channel location file formats
readlocs(): 'elc' format assumed from file extension
Reading file (lines): 10 20 30 40 50 60 70 80 90 100 110 120 130 140 150 160 170 180 190 200 210 220 230 240 250 260 270 280 290 300 310 320 330 340 350 360 370 380 390 400 410 420 430 440 450 460 470 480 490 500 510 520 530 540 550 560 570 580 590 600 610 620 630 640 650 660 670 680 690 700 702
3 channel with the same label found and imported
Channel lookup: no location for P3-Pz,C3-Pz,F3-Pz,Fz-Pz,F4-Pz,C4-Pz,P4-Pz,Cz-Pz,CM,A1-Pz,Fp1-Pz,Fp2-Pz,T3-Pz,T5-Pz,O1-Pz,O2-Pz,X3-Pz,X2-Pz,F7-Pz,F8-Pz,X1-Pz,A2-Pz,T6-Pz,T4-Pzreadlocs(): 'loc' format assumed from file extension

######################## Error in Red ###############

Brace indexing is not supported for variables of this type.

Error in pop_chanedit (line 1033)
    userdata.commands = { userdata.commands{:} args{:} };
 
Error while evaluating UIControl Callback.

####################

<br>
#### Steps to Reproduce

1. Import the eeg file (.edf)
2. Edit -> Channel Locs -> Default (Ok) - > Read Locs
3. import channel location file from sample channel location folder ( Standard-10-20-Cap25.locs)
4. autodetect -> ok
5. error (Can't avoid this erro)
6. look up locs
7. Use spherical file with eye channels (Can't find this option) -> ok

<br>
#### Expected behavior:
Channel locs get imported

<br>
#### Actual behavior:
getting error

<br>
#### Versions

Extensions Versions

EEGLAB: adding "Biosig" v3.8.4 to the path
EEGLAB: adding "ERPLAB" v12.00 (see >> help eegplugin_erplab)
EEGLAB: adding "ICLabel" v1.4 (see >> help eegplugin_iclabel) - new version 1.6 available
EEGLAB: adding "clean_rawdata" v2.8 (see >> help eegplugin_clean_rawdata) - new version 2.91 available
EEGLAB: adding "dipfit" v5.0 (see >> help eegplugin_dipfit) - new version 5.4 available
EEGLAB: adding "firfilt" v2.7.1 (see >> help eegplugin_firfilt) - new version 2.8 available

|   |   |
|----| ---- | 
| OS version     | [Windows 11] |
| Matlab version | [R2023a- academic use |
| EEGLAB version | [v2023.0] |

arnodelorme, 2025-01-07:
Can you share your EDF file?

arnodelorme, 2025-01-23:
Closing for now. Feel free to reopen.

closed 2025-01-23 by arnodelorme