sccn/eeglab #835: Error with LIMO: Unrecognized function or variable 'addons'

sbwinward, 2025-01-07:
### Description

When attempting to run the first level of the LIMO toolbox within EEGLab, I get the following error in a popup window:

"EEGLAB error in function limo_batch() at line 570:
Unrecognized function or variable 'addons'."

I don't understand why I'm getting this error as the addon that LIMO needs based on line 570 in LIMO (the parallel processing toolbox) is in my pathdef, as is the rest of the addons folder.The folder with my data and study file is linked below. 

[image]

[https://uofwaterloo-my.sharepoint.com/:f:/g/personal/swinward_uwaterloo_ca/EsAnDmccLyxBrQ4qjbXvbIsBS4coty-vQeYicw3fEoJ9tw?e=NdfLpM](url)

<br>
#### Steps to Reproduce

1. Open Matlab 2022b
2. Open EEGLab 2024.2.1 (with LIMO 3.4 as a plugin)
3. Load the .study file in the attached data folder ("EEG_Task_Study.study")
4. Select Study>LInear MOdelling of EEG Data>Estimate Model Parameters (channel)
5. Click "OK" on the popup
6. Wait some time for first level to run, it will produce the bug after a while.

<br>
#### Expected behavior:
I expected the first level of LIMO to continue running and eventually end by creating all the files necessary for the second level of analysis, along with the batch report.

<br>
#### Actual behavior:
The first level stopped running, I received the popup error message described above, and there is no batch report. At least some of the files for second level analysis were created though.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 11] |
| Matlab version | [2022b] |
| EEGLAB version | [2024.2.1] |
| LIMO version | [3.4] |

arnodelorme, 2025-01-23:
@CPernet do you know what the problem could be?

CPernet, 2025-01-23:
I have no idea. I recommend using v4.0 that we have now for a while and is way more stable.

arnodelorme, 2025-01-23:
Closing for now. Feel free to reopen the issue.

closed 2025-01-23 by arnodelorme