sccn/eeglab #837: EEG.event is empty after calling clean_artifact with channels_ignore

leima1, 2025-01-15:
### Description

[*EEG.event returns empty when calling clean_artifacts() with channels_ignore*]

On line 225 (+/- 1) of clean_artifacts function, EEG.event is emptied (EEG.event = []). 

It appears that Events get added back only if EEG = oriEEG gets called on line 345. However, line 345 only gets called if line 322 `~isempty(EEG.event) && isfield(EEG.event, 'type') && isstr(EEG.event(1).type)` is true. In situations where there is no event added to EEG between line 225 and line 322, then the original events never get added back. 

<br>
#### Steps to Reproduce

1. [*Set Highpass, BurstCriterion, WindowCriterion, and BurstRejection off, while passing a channel to ignore*]
EEG = pop_clean_rawdata(EEG, ...
        'FlatlineCriterion',5,'ChannelCriterion',0.7, ...
        'LineNoiseCriterion',4,'Highpass','off', ...
        'BurstCriterion','off','WindowCriterion','off', ...
        'BurstRejection','off','Distance','Euclidian', ...
        'channels_ignore',{'5Z'},'fusechanrej',1)

<br>
#### Expected behavior:

EEG.event should be the same as what it was passed into pop_clean_rawdata. I assume the ignored channel also gets placed back. 

<br>
#### Actual behavior:

EEG.event is empty and the channel that was ignored never gets placed back. 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 11*] |
| Matlab version | [*2023b*] |
| EEGLAB version | [*eeglab2023.1*] |
| clean_rawdata version | [*2.91*] |

arnodelorme, 2025-01-23:
Apologies for the late response. I do not see the same code on the lines you mentioned. 

https://github.com/sccn/clean_rawdata/blob/master/clean_artifacts.m

I am having a hard time understanding the issue. Is your issue that your events are numerical and this section of the code is not called?

leima1, 2025-01-23:
Hi Arnold!

Thanks for looking into this. 

Say I have an EEG structure with events, and the online reference channel is labeled as '5Z.' I want to remove noisy channels using the following line of code while making sure that '5Z' is ignored (i.e., not get removed).

```
EEG = pop_clean_rawdata(EEG, ...
    'FlatlineCriterion', 5, 'ChannelCriterion', 0.7, ...
    'LineNoiseCriterion', 4, 'Highpass', 'off', ...
    'BurstCriterion', 'off', 'WindowCriterion', 'off', ...
    'BurstRejection', 'off', 'Distance', 'Euclidian', ...
    'channels_ignore', {'5Z'}, 'fusechanrej', 1);
```

However, when I run this code, the returned EEG structure has an empty **EEG.event** field, and the channel '5Z' is removed. I would have expected that including _**'channels_ignore', {'5Z'}**_ was to retain that channel.

If I call the function without including **_'channels_ignore', {'5Z'}_** as an input, the **EEG.event** struct keeps its initial events. ‘5Z’ gets removed regardless. 

```
EEG = pop_clean_rawdata(EEG, ...
    'FlatlineCriterion', 5, 'ChannelCriterion', 0.7, ...
    'LineNoiseCriterion', 4, 'Highpass', 'off', ...
    'BurstCriterion', 'off', 'WindowCriterion', 'off', ...
    'BurstRejection', 'off', 'Distance', 'Euclidian', ...
    'fusechanrej', 1);
```

I will send you an example .set file where this issue occurs.

arnodelorme, 2025-01-23:
OK, that makes sense. Let me look into that.

arnodelorme, 2025-01-23:
Note for self: data downloaded in bug837 folder.

closed 2025-01-23 by arnodelorme