sccn/eeglab #838: Error in std_precomp

markpayumo, 2025-01-16:
### Description

Hi,

I'm trying to create ERSPs from STUDY without baseline correction. From my understanding, this can't be done on the GUI as it baseline corrects at default, so I need a script to run the precomputations.
This is the current script i have:

```
[STUDY, ALLEEG, customRes] = std_precomp(STUDY, ALLEEG, 'channels', {'all'}, ...
    'ersp', 'on', ...
    'erspparams', {'cycles', [13 0.1], 'nfreqs', 100, 'freqs', [13 30], 'freqscale', 'linear', 'ntimesout', 100}, ...
    'baseline', NaN);

```
However, when I try to run this, I get this error:

[image]

Even when I simplify the code like using a single channel or removing the parameters, I get this error.
How else can I get ERSPs that aren't baseline-corrected or get this code to work?
Thanks

<br>
#### Steps to Reproduce

1. Load STUDY in EEGLab
2. Run above std_precomp script

<br>
#### Expected behavior:
I expected it to run precomputations 

<br>
#### Actual behavior:
Gives me error

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Sequoia 15.1.1 |
| Matlab version | 2024b |
| EEGLAB version | EEGLab v2024.2.1 |

markpayumo, 2025-01-20:
I figured it out. The script doesn't like the channel array input for some reason. So this worked:

`[STUDY ALLEEG customRes] = std_precomp(STUDY, ALLEEG, {}, 'ersp', 'on', 'recompute', 'on','erspparams', {'cycles' [13 0.1] 'nfreqs' 100 'freqs' [10 30] 'ntimesout' 100 'freqscale' 'linear' 'baseline' NaN}) `

closed 2025-01-20 by markpayumo

arnodelorme, 2025-01-20:
Great