sccn/eeglab #839: Can't select mutiple Channel numbers in Channel Time Frequency --pop_newtimef()

saptrshighosh, 2025-01-16:
### Description

I am unable to select multiple channel numbers in Channel Time Frequency , even though it shows in the box that you can select multiple options pressing shift/ctrl.

I want to plot multiple channel Time Frequency using this Tool. But it limits me to one at a time.

https://www.dropbox.com/scl/fo/fa4tprbo1zpjjw47xlm15/ALoDv_rnX0uHnMQQVXEwS-E?rlkey=z5dzcyeseb2t4daln489p5xmj&st=p0t6wb3y&dl=0

<br>
#### Steps to Reproduce

1. *Upload an existing dataset*
2. *Plot -> Channel Time Frequency*
3. *Channel Number Box [...]*
4. *Unable to select multiple Channels*

<br>
#### Expected behavior:
A: Will be able to select multiple channels

<br>
#### Actual behavior:
A: Was not able to select multiple channels

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Window 11 Home*] |
| Matlab version | [*MATLAB R2023a - academic use*] |
| EEGLAB version | [*EEGLAB v2024.2.1*] |

closed 2025-01-23 by arnodelorme

saptrshighosh, 2025-01-23:
Thank you for the update! With the changes to the graphic interface, do we also need to update the extension? Additionally, for time-frequency analysis, how can I generate Time Frequency scalp plots for all channels?

arnodelorme, 2025-01-23:
For this, you need to create a study. https://eeglab.org/tutorials/10_Group_analysis/study_creation.html

saptrshighosh, 2025-01-23:
Thank you for your response. It seems I was on the right track, but I encountered an issue with the study design for time-frequency analysis. Should I create a new "Issue," or would you prefer I continue in this thread?

arnodelorme, 2025-01-23:
Yes, new issue. Tx.