sccn/eeglab #840: internal to std_makedesign inconsistent number of output args

dnacombo, 2025-01-17:
### Description

Running code from https://github.com/LIMO-EEG-Toolbox/limo_meeg/tree/master/resources/from_bids2stats.m
from this paper https://www.frontiersin.org/journals/neuroscience/articles/10.3389/fnins.2020.610388/full#figure2

```
[STUDY, ALLEEG] = pop_importbids(datapath,'bidsevent','on','bidschanloc','on',...
                  'studyName','Face_detection','outputdir', fullfile(datapath, 'derivatives'), ...
                  'eventtype', 'trial_type');

```
using full dataset stored locally.

returns error:

```
Error using std_getindvar
Too many output arguments.
Error in std_makedesign (line 187)
[indvars, indvarvals, ~, paired] = std_getindvar(STUDY);
                                   ^^^^^^^^^^^^^^^^^^^^
Error in std_checkset (line 204)
        STUDY  = std_makedesign(STUDY, ALLEEG);
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in std_editset (line 409)
[STUDY, ALLEEG] = std_checkset(STUDY, ALLEEG);
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in pop_importbids (line 700)
        [STUDY, ALLEEG]  = std_editset([], [], 'commands', commands, 'filename', studyName, 'task', task);
```
Seems like the code on line 187 uses a previous syntax to std_getindvar, expecting an output "paired" that isn't computed anymore.

All code to latest version.
I cannot fix.

arnodelorme, 2025-01-23:
Apologies for the late answer. It works for me, and I am not sure how to interpret the error, even after checking the std_getindvar function, which does have 4 output arguments.

Would you mind trying 

edit std_getindvar

You should have the same code as below with paired set to empty by default.

[image]

dnacombo, 2025-01-23:
My bad! sorry. I had an older local copy of that function on my computer; Sorry!

closed 2025-01-23 by dnacombo

arnodelorme, 2025-01-23:
no prob