sccn/eeglab #842: clarification of std_stat usage and interpretation of the outputs

leima1, 2025-01-22:
### Description

This is more of a question regarding the **pinter** output in the `std_stat `documentation, rather than a bug report. 

`function [pcond, pgroup, pinter, statscond, statsgroup, statsinter] = std_stat(data, varargin)
`

The example for how to organize the _data_ for `std_stat` in the eeglab documentation is as follows:
```
           >> data = { [800x12] [800x12] [800x12];... % 3 groups, cond 1
                        [800x12] [800x12] [800x12] };  % 3 groups, cond 2

```

And the outputs descriptions are:
```
% Outputs:
%  pcond        - [cell] condition pvalues or mask (0 or 1) if an alpha value
%                 is selected. One element per group.
%  pgroup       - [cell] group pvalues or mask (0 or 1). One element per 
%                 condition.
%  pinter       - [cell] three elements, condition pvalues (group pooled),
%                 group pvalues (condition pooled) and interaction pvalues.

```

If I understand correctly, **pinter{1}** provides p-values for the condition effect with groups pooled together, while **pinter{2}** provides p-values for the group effect with conditions pooled together. However, upon reviewing the outputs for **pinter**, it seems that the cells might be flipped, or the dimensions of the input data should be reversed.

<br>

### Steps to Reproduce
  
  
In the following code, I created dummy data for a 3 group (G) by 2 condition (C) and made sure that condition 2 (C2) is consistently larger than condition 1 (C1). I would expected that there is a main condition effect and no main group effect. However the pvalues from **pinter{1}** (i.e., condition) are 1, while **pinter{2}** (i.e., group) are on the magnitude of 1.0e-06 *.  

Nonetheless, when examining **pcond** and **pgroup**, the outputs appear as expected. For **pcond**, there are three cells, each showing very low p-values suggesting condition differences within each of the three groups. On the other hand, **pgroup** shows p-values of 1 for two cells, suggesting that there are no group differences within each condition.

I just want to make sure I am extracting and interpreting the stats correctly. 

Thank you.

```
G1C1 = rand(800,12);
G2C1 = rand(800,12);
G3C1 = rand(800,12);

G1C2 = rand(800,12)+2;
G2C2 = rand(800,12)+2;
G3C2 = rand(800,12)+2;

[pcond, pgroup, pinter, statscond, statsgroup, statsinter] ...
    = std_stat( ...
    {   G1C1 G2C1 G3C1;...
        G1C2 G2C2 G3C2},...
    'groupstats','on',...
    'condstats','on',...
    'method','permutation', ...
    'naccu',200,...
    'mcorrect','fdr');

disp('condition p-values')
pinter{1}
disp('group p-values')
pinter{2}

% visualize
p1 = plot(pinter{1},'r',LineWidth=2);hold on
p2 = plot(pinter{2},'b',LineWidth=2);
yline(0.05,'--',LineWidth=2)
legend([p1 p2],{'condition p-values','group p-values'})

```

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 11*] |
| Matlab version | [*2023b*] |
| EEGLAB version | [*2024.2*] |

arnodelorme, 2025-01-23:
Yes, you are right. I have run your code and the outputs for pinter is flipped. I have corrected the function help message. Thank you for reporting.

closed 2025-01-23 by arnodelorme

leima1, 2025-06-28:
Hi @arnodelorme . Thanks for updating, I see that the `pinter `description is updated, but is the corresponding `statinter` description updated as well? I just want to double check and make sure I am (and others are) interpreting the outputs correctly. Thank you.