sccn/eeglab #844: ERP Lab > ERP Measurement Tool > Saved Output file extension bug

saptrshighosh, 2025-01-26:
### There is a bug in ERPLab's ERP Measurement Tool where the output file is, by default, saved in the .xls format, which is a spreadsheet file. However, EEGLAB no longer supports this format and can only save files in the .txt format. As a result, users must manually change the file extension to .txt while saving the file through the GUI to their desired location.There is a bug in ERPLab's ERP Measurement Tool where the output file is, by default, saved in the .xls format, which is a spreadsheet file. However, EEGLAB no longer supports this format and can only save files in the .txt format. As a result, users must manually change the file extension to .txt while saving the file through the GUI to their desired location.

https://www.dropbox.com/scl/fo/fa4tprbo1zpjjw47xlm15/ALoDv_rnX0uHnMQQVXEwS-E?rlkey=aghiypjjmrocn8h4z5zjzm0iz&st=i17spct7&dl=0

<br>
#### Steps to Reproduce

1. Post ICA
2. Create EEG event list
3. Make BDF 
4. Extract Bin based epochs
5. Moving window peak to peak threshold ->90
6. Computer Avg erps
7. Filters for ERP data  - 30 hz
8. ERP channel Operations
9. ERP measurement Tool
10. Type = Mean Amplitude, Bins and Channel = Default, Measurement Window =  300 500
11. RUN

<br>
#### Expected behavior:
would directly save as a .txt file

<br>
#### Actual behavior:
tries to save in .xls with error

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Window 11 Home*] |
| Matlab version | [*MATLAB R2023a - academic use*] |
| EEGLAB version | [*EEGLAB v2024.2.1*] |

arnodelorme, 2025-01-26:
This issue pertains to EEGLAB, not ERPLAB, on this website. It is recommended to report this problem at https://github.com/ucdavis/erplab/issues.

closed 2025-01-26 by arnodelorme

saptrshighosh, 2025-01-26:
Apologies for that.

https://github.com/ucdavis/erplab/issues/201