sccn/eeglab #845: std_precomp Error When inds Becomes Empty

sato16, 2025-01-29:
Dear EEGLAB developers,

### Description

When loading data such as P1_sess1, P1_sess2, P2_sess1, P2_sess2 and running std_precomp, the variable inds inside the following nested loop becomes empty:

```
for iSubj = 1:length(uniqueSubjects)
    for iSess = 1:length(uniqueSessions)
        inds1 = strmatch(uniqueSubjects{iSubj}, allSubjects, 'exact');
        inds2 = strmatch(uniqueSessions{iSess}, allSessions, 'exact');
        inds  = intersect(inds1, inds2);
    end
end
```

In line 345 of `compute spectrum`, there is a condition `if ~isempty(inds)`, but I did not find a similar check in lines 397 and 453. It seems that the same handling may be necessary in those lines as well.

Code:
```
 [STUDY, ALLEEG] = std_precomp(STUDY, ALLEEG, {},'savetrials','on',...
            'interp','on',...
            'recompute','on',...
            'erp','on','erpparams',{'rmbase',[-1000 0] },...
            'spec','on','specparams',{'specmode','fft','logtrials','off'},...
            'erpim','on','erpimparams',{'nlines',10,'smoothing',10},...
            'ersp','on','erspparams',{'cycles',[3 0.8] ,'nfreqs',100,'ntimesout',200},...
            'itc','on');
```

Error message:
Index exceeds the number of array elements. Index must not exceed 0.
- Error in std_precomp (line 397)
  filepath = STUDY.datasetinfo(inds(1)).filepath;

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*macOS Monterey*] |
| Matlab version | [*2024 a*] |
| EEGLAB version | [*2024.2*] |

Thank you for your support.
Best regards,
Fumiaki

sato16, 2025-01-30:
Sorry for posting again.
I have realized that the subject names for sess1 and sess2 should not be different. When using the same subject name across sessions, the function appears to work correctly.

However, if one of the sessions is missing for some reason, the function may not work as expected. To handle this case properly, it seems necessary to include a condition such as if ~isempty(inds), or at least a warning message to alert the user.

Thank you for your consideration.

arnodelorme, 2025-02-06:
That's a good point. I put the fix in 5139a81a9
Let me know if that works for you.

closed 2025-02-06 by arnodelorme