sccn/eeglab #846: Question About Exported Data Units in EEGLAB v6.03b

23111507, 2025-01-29:
I am currently using EEGLAB v6.03b for EEG data processing and have encountered an issue regarding the unit of the exported data when analyzing absolute alpha power. I would appreciate guidance in confirming the correct unit and whether any transformations are required to match our expected format.
The data that we extracted:
0,0339722	0,10060299	0,05050097	0,01392906	0,01628528	0,01611051	0,03875389	0,0513764	0,00673493	0,14145665	0,03786448	0,00757985	0,0406878	0,01536764	0,02309678	0,03341658	0,05817799	0,05812679	0,03116332	0,01769405
0,01775447	0,03355406	0,01447672	0,02947469	0,02612831	0,04012568	0,02516946	0,02964078	0,03552746	0,0219075	0,03619714	0,02085587	0,0302219	0,03095064	0,02536449	0,0336114	0,01653505	0,01928295	0,01920426	0,02102296
0,01623331	0,0356797	0,01388128	0,02581813	0,02717595	

I would like to confirm whether the exported absolute alpha power values are in Volts (V) or microvolts squared (µV²). Our expected format requires the unit to be µV², and we need to ensure that we are interpreting the data correctly. The unit that we expected was like this example:

Fp1 pré | Fp1 pós | Fp1 pós 30 | Fp2 pré | Fp2  pós | Fp2 pós 30

3,959671 | 2,264832 | 2,6023268 | 3,931522 | 2,449325 | 2,96393233
6,913344 | 7,642714 | 8,6252414 | 8,266277 | 9,789824 | 9,63509941
12,82466 | 11,45018 | 14,331688 | 13,83497 | 11,51249 | 14,6532677
9,57815 | 8,386484 | 5,0466436 | 10,31061 | 8,823149 | 5,37831897
8,144927 | 4,919925 | 8,9143992 | 7,814815 | 6,051634 | 9,96923938
27,33856 | 19,55958 | 27,834536 | 32,65109 | 22,5823 | 31,8768957

Additionally, we would appreciate guidance on where in the software settings we can modify the unit to match our expected format.

Questions:
What is the default unit of measurement in EEGLAB v6.03b for absolute power values?
If the data is stored in Volts (V), what is the correct transformation to obtain µV²?
Where in the EEGLAB interface or MATLAB scripts can we confirm and modify this setting?

arnodelorme, 2025-02-03:
The exported alpha power values are definitely not in Volts or microVolts, they are either in microVolt^2 or 10*log10(microVolt^2).

closed 2025-02-03 by arnodelorme

23111507, 2025-02-05:
Thank you for your response. I understand that the exported alpha power values are either in µV² or 10 × log10(µV²). However, I am still unsure about how to properly interpret my extracted values and convert them to match our expected format.

How can I determine whether my exported values are in µV² or 10 × log10(µV²)? Is there a way to verify this within EEGLAB or MATLAB?
If my values are in 10 × log10(µV²), what is the exact transformation to convert them back to µV²?
Should I apply the formula 10^(value/10) to each data point?
I expected my absolute alpha power values to be within a range like [0, 90], but my extracted values are between [0.01, 0.25].
Does EEGLAB apply any additional scaling that could explain this discrepancy?
If so, how can I adjust my results accordingly?
Additionally, could you point me to where in EEGLAB (or its scripts) the power computation settings are stored, so I can confirm the default unit being used?

I appreciate your time and guidance in clarifying this issue.

arnodelorme, 2025-02-05:
You can plot the spectrum and you will see right away if it is log power or not. This is log power spectrum

https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcSE0ClVcoqNAhcJ7hMBcRsB4wmvCmaa8jIKCg&s

If it goes flat right away though it is microvolt square. To go back from log power spectrum to microvolt square, you do 10^(x/10)