sccn/eeglab #847: Issue with exporting to Ragu in MICROSTATELAB

Nuggetsnsauce, 2025-01-31:
### Description

Hi there, I am having trouble exporting my data to Ragu to test for topographical differences in microstates in microstatelab. Everytime I get to the end and try to do it I get an error saying this: 

https://docs.google.com/document/d/1WglagaJ6tUTyr2I_rx5d5C-qgkFRlUaYEuGFOkj1NcU/edit?usp=sharing 

<br>
#### Steps to Reproduce

I just follow the MICROSTATELAB Tutorial with my dataset and get this issue whether I am using all 59 of the sets or just 6. I will attatch 6 for you to see (3 ADHD and 3 controls - these are the two groups I am trying to compare). I will also attatch the microstatelab tutorial for you to see.

Here is the link to access the data:
https://www.dropbox.com/scl/fo/8iq6gljb93kdb9kdusrus/ABRCCXBnh0IuNMwgIkcfyiw?rlkey=j74hyoqjpx9hwn4h86iid10n5&st=6z2pb2s0&dl=0

Here is the link to the Microstatelab tutorial:
https://link.springer.com/article/10.1007/s10548-023-01003-5

<br>
#### Expected behavior:
I should be able to use the pop-up window to define between subjects groups and then compare the maps.

<br>
#### Actual behavior:
I get the error.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [22H2] |
| Matlab version | [R2024b] |
| EEGLAB version | [2024.2.1] |

arnodelorme, 2025-01-31:
This has to do with an EEGLAB plugin. I would contact the plugin authors directly.

closed 2025-01-31 by arnodelorme

Nuggetsnsauce, 2025-01-31:
Okay, thank you :)