sccn/eeglab #850: Issue with study stats

arnodelorme, 2025-02-11:
I'm exploring STUDY statistics for the first time in a 2x2 factorial design. I have sequentially compared "Compute 1st independent variable..." and "Compute 2nd independent variable...", using EEGLAB's permutation/Bonferroni option (n=1000, threshold = .05) and using Fieldtrip statistics (montecarlo, cluster correction, n=1000, threshold = .05).
The results are almost impossible to reconcile, unless what the EEGLAB statistics consider the 1st independent variable is the 2nd one for the Fieldtrip statistics option and vice versa.

arnodelorme, 2025-02-11:
I am confused, because the fieldtrip option cannot compute 2x2 designs, only 1xm designs.

KrisBaetens, 2025-02-12:
I understand, because I might not have expressed myself very clearly :). So, I have a 2x2 design, manipulating Go/NoGo, and alcohol versus non-alcohol related context. In all the below examples, I show scalp maps in the 300-500 ms. As one can readily observe, there is a robust Go/NoGo effect, and maybe, maybe not, a subtle influence of alcohol context. So, running: 

**EEGLAB stats 1st factor (only), bonferroni, threshold .01, n = 1000**, you get: 
[image]
So, no effect here. Alright, so I take the title of the right plot to mean "(Collapsed across) Go/NoGo", there not being much of an effect of alcohol context. That would be a counter-intuitive labeling of the figure imho (it would be more straightforward if GoNoGo displays the effect of the factor Go/NoGo), but it is not impossible.

**Fieldtrip stats 1st factor (only) Fieldtrip (monte carlo, .01, n = 1000, cluster correction), one gets:**
[image]
At first I thought, okay, the cluster correction is probably more lenient, makes sense, here an effect of alcohol _is_ detected. So, progressing with otherwise identical parameters, but moving on to the 2nd independent (only),

**EEGLAB 2nd factor(only):**
[image]
So, as is immediately apparent, this looks very similar to the fieldtrip results for the first factor, previous plot. 

**Fieldtrip 2nd factor (only):**
[image]
There is an effect here, which is not present in the EEGLAB stats for the 1st independent. However, if we make the threshold in EEGLAB stats more lenient, we get something more similar to this for the **1st independent**:
[image]
From this, it would seem that:
- the 1st independent using EEGLAB stats = alcohol vs no alcohol
- the 1st independent using Fieldtrip stats = go vs no go
- the inverse is true for the 2nd independent.

So, I conclude two things from this:
- In a 2x2 design, **if you calculate the effect of only 1 independent, the actual factor used is swapped between the EEGLAB and Fieldtrip statistics option** - which seems undesirable, if true.
- The labeling of the Fieldtrip stats corresponds more to intuition: GoNoGo shows you the effect of this factor, not the effect of the other factor "when you collapse across the one mentioned in the title". Such a way of naming things makes sense in the marginal effects plot:
[image]
but can lead to critically erroneous conclusions when only considering the interaction plot:
[image]

arnodelorme, 2025-02-12:
I think the real way to compare is to use the same stats. This is for the first condition EEGLAB on the left and Fieldtrip on the right)

[image]

This is for the second condition. EEGLAB on the left and Fieldtrip on the right)

[image]

arnodelorme, 2025-02-12:
I am closing this report, but let me know if you have more questions.

arnodelorme, 2025-02-12:
Would you mind reposting with the images (they do not show up). Are you using the latest version of EEGLAB? There was a problem that was fixed a couple of months ago. Would you mind trying with the version below.

https://sccn.ucsd.edu/eeglab/currentversion/eeglab2025.0.0rc.zip

arnodelorme, 2025-02-13:
Unfortunately, I cannot see the word doc on Github. Can you send it to my personal email at arnodelorme at gmail dot com. Also, is there a way to access the data to reproduce the issue?