sccn/eeglab #855: Error in filtering using ERPLAB in EEGLAB

adwaitdagitguy, 2025-03-08:
I tried filtering the sample data in eeglab (2025.0.0) using the ERPLAB (12.00) function "Filters and Frequency Tools > Filters For EEG Data" and then I get the following error:
EEGLAB error in function plotresponsefilter() at line 1202:Unrecognized field name "axes1".If you think this is a bug, please submit a detailed description of how to reproduce the problem (and uploada small dataset) at https://github.com/sccn/eeglab/issues

I am using MATLAB online and the sample dataset eeglab_data.set

[image]

oskaosas, 2025-04-02:
Is it fixed?
facing the same problem here.

arnodelorme, 2025-04-03:
This is an ERPLAB problem. I have copied the problem to https://github.com/ucdavis/erplab/issues/206

davidg622, 2025-04-07:
Hi,

Our team tried to reproduce this issue using the example dataset from EEGLAB but were unsuccessful. Could you please share your MATLAB version and operating system?

Thanks,
David

davidg622, 2025-04-29:
Hi Adwait,

Thank you for bringing this to our attention.

After looking into the issue, it appears that the error occurs specifically when ERPLAB is run via MATLAB Online. Unfortunately, this is due to limitations in the apps used to launch our GUI, which are currently not supported in the online environment and fall outside our control.

At this time, we’re unable to offer support for GUI usage within MATLAB Online. We recommend running ERPLAB from a locally installed version of MATLAB for full functionality, or using scripts if you prefer to remain in the online environment.

We sincerely apologize for the inconvenience, and we appreciate your understanding.

Best regards,
David

adwaitdagitguy, 2025-05-04:
Hello David

Thanks for your clarification and for testing out this issue.

I used MATLAB online because the local MATLAB was crashing a lot on my system due to hardware constraints. Although there are some GUI limitations, I did some processing in EEGLAB/ERPLAB (of about 10 8-channeled files and 2-3 64-channeled files), using scripts as you mentioned, as an equivalent method. Maybe I should refer the scripting guide available in the ERPLAB repo (https://github.com/ucdavis/erplab/wiki/Scripting-Guide) for processing the data. Also, there are some features of ERPLAB which **do** work in the online GUI like Plot ERPs and same applies for EEGLAB.

Anyways, thanks for your time! I would like to keep you all updated on the full scripting experience in EEGLAB/ERPLAB in MATLAB online.

Best Regards
Adwait

closed 2025-09-10 by arnodelorme