sccn/eeglab #858: Incorrect EEG.times vector in epoched data (if epoch start time is after event)

olafdimigen, 2025-03-24:
### Description
In EEGLAB 2024.2, when cutting epochs, the EEG.times vector is incorrect if the epoch start is *after* the event and there is only one epoch of the given type.

In that case, while the size of EEG.data will be correct, the time vector in EEG.times remains that of the original, continuous dataset from which epochs were cut (i.e. the vector is incorrect). 

Example:

```
EEGepoched = pop_epoch(EEG,'s255', [-0.01 0.01]); 
length(EEGepoched.times)
% ans = 160 % (correct, this is data sampled at 8kHz)

EEGepoched = pop_epoch(EEG,'s255', [0.01 0.03]); 
length(EEGepoched.times)
% ans =  444353 (lenght of cont. dataset)
```

#### Steps to Reproduce
See above. Notably, in my dataset there is only a single epoch of each event type (here: "s255"), which likely causes the issue, as it is interpreted as still continuous data.

|----| ---- | 
OS version:  Win10
Matlab version: R2023b
EEGLAB version: 2024.2

arnodelorme, 2025-03-25:
Thanks @olafdimigen, I will let you know when this is addressed.

arnodelorme, 2025-04-03:
Hi @olafdimigen, this was fixed in 8074b6162
Let us know.

closed 2025-04-03 by arnodelorme