sccn/eeglab #859: Circular error using ICLabel in the GUI

ShekharNarayanan, 2025-03-31:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. Downsample EEG Data
2. Filter data
3. Decompose data by ICA
4. Classify components using ICLabel

If I click on "Flag components as artifacts" the cmd window tells me to classify the components first. When I try to classify the components first it still tells me to classify the errors first. I have no other issues filtering or running the ICA.

Error:
`No labeling found, use Tools > Classify components using ICLabel > Label components first`
<br>
#### Expected behavior:
I have tried the same step many times and usually a window with labelled components like eye, heart, muscle pops up. I have never had this circular error.

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     |Windows 11|
| Matlab version | 2023b |
| EEGLAB version | 2024.2 |

arnodelorme, 2025-04-03:
Did you try labeling components first? Would you mind sharing your dataset prior to calling ICLabel?

ShekharNarayanan, 2025-04-04:
Hi!

Thanks a lot for your reply! I was under the impression that EEGLAB labels the components automatically? This was my experience before. I have attached a screenshot of when EEGLAB labelled the components for me after following the same steps listed before.

[image]

Unfortunately, I am unable to share the dataset due to privacy and GDPR compliancy issues as I work with a university. 

I wish to have the labelling done automatically as the main users for the protocol I am building are students who are not subject matter experts. 

Would trying out a legacy EEGLAB version be a potential solution?

arnodelorme, 2025-04-04:
After computing the percentage, you have to select the next menu to flag the component (this is where you manually choose threshold for component classes).

closed 2025-04-04 by arnodelorme

ShekharNarayanan, 2025-04-08:
I am unable to label components at all. I am getting this message `No labeling found, use Tools > Classify components using ICLabel > Label components first` when I follow the exact steps mentioned in the error- meaning I click on Label components, but I end up getting the same error as I just mentioned.