sccn/eeglab #863: Error in EEGLAB useing LIMO when processing ersp data

qushiyan, 2025-04-23:
The version of EEGLAB I am using is 2022.0, the version of FieldTrip version is 2020, and my MATLAB version is 2022b. After I precomputed my spec and ersp, I edited my study design. I want to use the continuous variable PE (each trial corresponds to a PE value) to perform a regression analysis with my ersp and then use an independent samples t-test to identify the time-frequency points related to PE. I follow the instruction in eeglab.wiki, every thing goes well until I ran my code for ersp:

STUDY = pop_limo(STUDY, ALLEEG, 'method','WLS','measure','dattimef',...
    'freqlim',[2 50],'erase','on','splitreg','off','interaction','off');
However, it shows the following error: "Unrecognized field name 'F'." The error occurs at line 865 in the limo_glm function:
tmpF(f,:) = reshape(model.conditions.F(f,:,:), [n_freqs*n_times,1]);
But when I process ERP and spec data, there are no issues. I wonder where is the problem occured.

CPernet, 2025-06-12:
Please try limo master branch from GitHub - this should work. Let me know if not and we'll fix it. Thx 🙏

arnodelorme, 2025-06-12:
Is that the latest on the plugin manager? Should we release a new version?

CPernet, 2025-06-17:
let me fix some GUI stuff (all outputs now follow BIDS but 2nd level GUI does not yet found all files)

CPernet, 2025-06-17:
@qushiyan did this work?

qushiyan, 2025-06-17:
Unfortunately, it didn't work. I found that different versions of EEGLAB and LIMO have a significant impact on whether they can continue to run. I ultimately calculated beta and do statistical test by Matlab code directly.

发自我的iPhone

closed 2025-08-06 by CPernet