sccn/eeglab #865: clean_channels() / clean_artifacts() leads to different channel removed results

leima1, 2025-04-29:
### Description

This is more a minor suggestion than a bug report.  When running clean_channels() using parallel processing, each worker uses an independent random number stream. This causes reproducibility issues when re-running the analysis with parallel processing (i.e., the same worker might not be assigned to run the same subject or trial when clean_channels() called, leading to a different outcome for that subject or trial). 

According to https://www.mathworks.com/help/parallel-computing/control-random-number-streams-on-workers.html, setting rng(0, 'Philox'), instead of rng('default') ensures that all workers use the same random number generator. In our experience, replacing that line within clean_channels() has stabilized the channel removal results across repeated runs.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*2023b*] |
| EEGLAB version | [*2025*] |

chkothe, 2025-05-06:
@arnodelorme This is indeed related to something I had mentioned to you. An ancient version of `clean_channels` (and the one in BCILAB) used always a fixed seed for reproducibility reasons, but from what I can tell, that had already been disabled before the initial commit in the `clean_rawdata` repo. It does look like this was updated in '22 by using MATLAB's default seed, i.e., `rng('default')`, but as you noticed, the behavior of that might differ when used inside a parfor loop.

It would be desirable that this function continues to produce the same random seeds that it's been producing for the past few years so that people can reproduce their prior EEG studies, and also for compatibility with ports of this toolbox to other programming languages (which usually aim to produce bit-identical results to MATLAB to the extent possible) -- for this reason I would strongly suggest changing the line to:

```
rng(5489, 'mt19937ar');
```

which is MATLAB's default random number generator and the built-in default seed that it uses in the single-process setting, so that will recover the _same_ behavior as rng('default') and thus reproduce the same channel sets that clean_rawdata produced in the past few years.

You can check that by Comparung the output of:
```
rng('default'); disp(rand(10))
```
... versus a simulated parallel run:
```
parfor i=[1:3]; rng(5489, 'mt19937ar'); disp(rand(10)); end
```

... and a single-process run of the same:
```
rng(5489, 'mt19937ar'); disp(rand(10));
```
which should all print the same matrix.

Note that using the newer Philox RNG instead will NOT reproduce prior results.

arnodelorme, 2025-09-10:
@kothe, do we need to make any changes?