sccn/eeglab #867: bug in pop_chanedit

jiversen, 2025-05-08:
## Description

```pop_chanedit``` returns a malformed ```command``` into eegh. Running that command does not recreate the behaviour that was done in the gui.

### Steps to Reproduce:

1. I started ```pop_chanedit``` from the menu on a 7-channel dataset that has only 1020 channel labels , but no channel locations or channel types. [The main window lists channel locations *"No (labels only)"*]. 

      I picked the standard head model then set all of the channel types to 'EEG'--this properly populates the channels with all of the standard coordinate values and returns a dataset with full chanlocs. Fine.

2. However, I then took the command string stored in ```eegh``` and added it to my own code

3. Running this command on a new dataset _does not work_: it returns an ```EEG.chanlocs``` with empty coordinates!

### Expected behavior:

Running the code returned by ```pop_chanedit``` should properly lookup the electrode coordinates, and recreate the GUI behavior, not return empty coordinates.

### Actual behavior:

The command string that was returned to eegh for the action was malformed and does not run properly:

```EEG=pop_chanedit(EEG, {'lookup','/XXXX/eeglab/plugins/dipfit/standard_BEM/elec/standard_1005.elc'},'changefield',{1,'type','EEG'},'changefield',{2,'type','EEG'},'changefield',{3,'type','EEG'},'changefield',{4,'type','EEG'},'changefield',{5,'type','EEG'},'changefield',{6,'type','EEG'},'changefield',{7,'type','EEG'});```

Notice how the second argument is a _cell array_ -- this case is not handled in the pop_chanedit argument parsing code around line 248 (the case when orichaninfo is a cell is not handled, only the case when is a char).

**This leads to the lookup param/value pair to be completly ignored, so the returned EEG.chanlocs has no coordinate values.**

### Suggested fix:

Fix the code that builds the command string. It should return:

```EEG=pop_chanedit(EEG, 'lookup', '/XXXX/eeglab/plugins/dipfit/standard_BEM/elec/standard_1005.elc', 'changefield',{1,'type','EEG'},'changefield',{2,'type','EEG'},'changefield',{3,'type','EEG'},'changefield',{4,'type','EEG'},'changefield',{5,'type','EEG'},'changefield',{6,'type','EEG'},'changefield',{7,'type','EEG'});```

**Note there are no brackets around the lookup/path PV pair. This will be properly parsed by ```pop_chanedit```**

### Workaround
Manually reformat the command you copied from eegh to remove those curly brackets.

### Versions

|   |   |
|----| ---- | 
| OS version     | macos 15.4.1 |
| Matlab version | 2024b |
| EEGLAB version | 2025.0.0 |

arnodelorme, 2025-05-08:
I am pretty sure this was fixed in 2025.1. Would you mind checking?

jiversen, 2025-05-08:
Sorry, should have checked that first! Sure, I'll update.

arnodelorme, 2025-05-08:
Actually maybe use the GitHub version because the problem might have been fixed after 2025.1 (I can’t remember). I confirmed that it is fixed for me right now.