sccn/eeglab #868: pop_runica fails to find installed plugin AMICA due to faulty existence test

zenBen, 2025-05-17:
### Description

In EEGLAB dev, `pop_runica` tests for existence of files named as per the list of algorithm names (lines 157 - 161).
```
for index = length(allalgs):-1:1
    if ~exist(allalgs(index).name, 'file')
        allalgs(index) = [];
    end
end
```

- Algorithm names are given in the `allalgs` structure (lines 111 - 154), this contains about 29 algorithm names, including `amica`.

- AMICA 1.7 installed via plugin manager does not contain any file named `amica`.

- Thus, the `pop_runica` code above produces a list of available algorithms that doesn't match reality.

<br>
#### Steps to Reproduce

1. Install EEGLAB dev
2. Install AMICA 1.7 plugin
3. try to run ICA on a dataset using method AMICA

<br>
#### Expected behavior:
AMICA is performed on EEG data

<br>
#### Actual behavior:
`pop_runica` fails

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2025-06-12:
Sorry about that. We will look into the problem. Also, AMICA is usually run outside of this function (it has its own menu).

arnodelorme, 2025-09-10:
Fixed in e505b7121

closed 2025-09-10 by arnodelorme