sccn/eeglab #869: Unsuccesful_attempt to run 3 factor repeated ANOVA

mileverocarota, 2025-05-19:
### Description

[*It appears impossible to obtain a three factor ANOVA when using LIMO eeglab toolbox*]

Here I leave below two type of designs I tried forming using my data to run a three factor ANOVA. It seems impossible for me to succesfully construct a right design matrix in LIMO to conduct analyses using a 3 x 2 x 2 ANOVA analyses.

Description of variables/data:

2 x Groups of participants (OLD = EA and YOUNG = YA)

3 x Object at fixation 
      - Target (T)
      - Distractor 1 (D1, same colour)
      - Distractor 2 (D2, dif colour)

2 x Fixation Ranks
      - First fixation (F1)
      - Last fixation (F2)

2 x Types of search
      - Single search (S)
      - Dual Search (D)

<br>
#### Steps to Reproduce

1. [*Design 1 attempt, my code*]
CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];
STUDY = std_makedesign(STUDY, ALLEEG, designN, 'name','D3D_StxfR-fObj','delfiles','off', ...
    'defaultdesign','off', ...
    'variable1','group','values1',{'EA','YA'},'vartype1','categorical','variable2', ...
    'type','values2',{ ...
    'S_T_F1', 'S_T_FLast', 'D_T_F1', 'D_T_FLast', ...
    'S_D1_F1', 'S_D1_FLast', 'D_D1_F1', 'D_D1_FLast', ...
    'S_D2_F1', 'S_D2_FLast', 'D_D2_F1', 'D_D2_FLast'}, ...
    'vartype2','categorical','subjselect',{'EA20','EA21','EA22', ...
    'EA23','EA24','EA25','EA26','EA27','EA28','EA29','EA30','EA31','EA32','EA33','EA35','EA36', ...
    'EA37','EA38','EA39','EA40','EA41','EA42','EA43','EA44','EA45','YA21','YA22','YA23','YA24', ...
    'YA25','YA26','YA27','YA28','YA29','YA30','YA31','YA32','YA33','YA34','YA35','YA36','YA38', ...
    'YA39','YA40','YA42','YA44'});

STUDY = std_selectdesign(STUDY, ALLEEG, designN);
eeglab redraw;

2. [*Design 2 attempt, my code*]

CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];
STUDY = std_makedesign(STUDY, ALLEEG, designN, 'name','D3_Finale','delfiles','off', ...
    'defaultdesign','off', ...
    'variable1','group','values1',{'EA','YA'},'vartype1','categorical','variable2', ...
    'type','values2',{ ...
    {'D_T_F1', 'D_D1_F1', 'D_D2_F1'}, ...          % Dual & F1 con_1
    {'D_T_FLast', 'D_D1_FLast', 'D_D2_FLast'}, ... % Dual & FLast con_2
    {'S_T_F1', 'S_D1_F1', 'S_D2_F1'}, ...          % Single & F1 con_3
    {'S_T_FLast', 'S_D1_FLast', 'S_D2_FLast'} ...  % Single & FLast con_4
    }, ...
    'vartype2','categorical','subjselect',{'EA20','EA21','EA22', ...
    'EA23','EA24','EA25','EA26','EA27','EA28','EA29','EA30','EA31','EA32','EA33','EA35','EA36', ...
    'EA37','EA38','EA39','EA40','EA41','EA42','EA43','EA44','EA45','YA21','YA22','YA23','YA24', ...
    'YA25','YA26','YA27','YA28','YA29','YA30','YA31','YA32','YA33','YA34','YA35','YA36','YA38', ...
    'YA39','YA40','YA42','YA44'});

STUDY = std_selectdesign(STUDY, ALLEEG, designN);
eeglab redraw;

% ----------------------------------
%% Save your Study with 3rd design
% ----------------------------------

[STUDY EEG] = pop_savestudy( STUDY, EEG, 'filename','D3_Finale_fObj-Search-Frank.study', ...
    'filepath', StudyFold);
CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];
eeglab redraw;

% ----------------------------------------------------------------
%%  Precompute ERP and Spectrum measures - 
% these can also be computed after designing study design
% ----------------------------------------------------------------

% % ---------------------------------
% % re-load previous design study
% % ---------------------------------
% [ALLEEG EEG CURRENTSET ALLCOM] = eeglab;
% 
% [STUDY ALLEEG] = pop_loadstudy('filename', 'D3_fObj-Search-Frank.study', 'filepath', ...
% StudyFold);
% 
% CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];
% 
% % you may need to use this code if study.trials are inconsistent with
% % ALLEEG.trials

% if this study fails to process when using a third design we can instead
% load study T2b and create a new design (1) with fRank x f_Obj

designN = 3; % this is the third model   

[STUDY, EEG] = std_precomp(STUDY, ALLEEG, {}, 'savetrials','on','interp','on','recompute','on',...
    'erp','on','erpparams', {'rmbase' [-200 -100]}, 'spec','off',...
    'ersp','off','itc','off', 'specparams',{'specmode','fft','logtrials','off'});

% ------------------------------------------------
% Compute GLM (wait 10mins)
% ------------------------------------------------

% addpath('C:\Users\local user\Documents\MATLAB\Toolboxes\fieldtrip-20250106\connectivity\private')
STUDY  = pop_limo(STUDY, EEG, 'method','WLS','measure','daterp','timelim',[-200 300],...
    'erase','on','splitreg','off','interaction','on');

% --------------------------------------------------------
%% Refine strucuture in Study.limo (change at every model)
% before computing analyses
% --------------------------------------------------------
n = length(STUDY.limo.betas);
nFact = n - 1 %(constant);

% first three are from D1
start = 3;

for i = start + 1:nFact
    if i >= 4 && i <= 6
    STUDY.limo.betas(i).label = 'Dual- F1'
    elseif i >= 7 && i <= 9
    STUDY.limo.betas(i).label = 'Dual- FLast'
    elseif i >= 10 && i <= 12
    STUDY.limo.betas(i).label = 'Single - F1'
    else 
    STUDY.limo.betas(i).label = 'Single - FLast'
    end
end

% levels 

for i = start + 1:nFact
    if i >= 4 && i <= 6
    STUDY.limo.betas(i).level = 'one'
    elseif i >= 7 && i <= 9
    STUDY.limo.betas(i).level = 'two'
    elseif i >= 10 && i <= 12
    STUDY.limo.betas(i).level = 'one'
    else 
    STUDY.limo.betas(i).level = 'two'
    end
end

for i = start + 1:nFact
    if i >= 4 && i <= 6
    STUDY.limo.betas(i).description = 'Dual- F1'
    elseif i >= 7 && i <= 9
    STUDY.limo.betas(i).description = 'Dual- FLast'
    elseif i >= 10 && i <= 12
    STUDY.limo.betas(i).description = 'Single - F1'
    else 
    STUDY.limo.betas(i).description = 'Single - FLast'
    end
end

% for i = 1:length(STUDY.limo.betas)
%     STUDY.limo.betas(i).description = [STUDY.limo.betas(i).value]
% end

% ------------------------------------------------
%% Compute GLM (wait 10mins)
% ------------------------------------------------

STUDY  = pop_limo(STUDY, EEG, 'method','WLS','measure','daterp','timelim',[-200 300],...
    'erase','on','splitreg','off','interaction','on');

% ----------------------------------------------------
% Save your Study with 1st design + Precomputations
% ----------------------------------------------------

[STUDY EEG] = pop_savestudy( STUDY, EEG, 'filename','D3_fObj-Search-Frank.study', ...
    'filepath', StudyFold);
CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];
eeglab redraw;

3. [*and so on...*]

<br>
#### Expected behavior:
[*The ANOVA is running but I cannot compute a three factor ANOVA. My outputs are limited to my fiirst three objects *]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*MATLAB 2024b*] |
| EEGLAB version | [*EEGLAB 2025_01_06*] |

arnodelorme, 2025-05-19:
Can you share the data with @CPernet ?

mileverocarota, 2025-05-19:
Happy to share my data can you advice me how?

arnodelorme, 2025-06-12:
Any link to Google Drive would work. Feel free to drag and drop data here https://drive.google.com/drive/folders/1RWa6UuvPEzWlF6aSMWvd3pagl82lH92K?usp=sharing

@CPernet can you look at the issue?

arnodelorme, 2025-06-16:
Ah, thank you, please let us know.

CPernet, 2025-06-22:
I updated the header of limo_random_select with an exemple (in master branch) - the way to do it is like explained: 

```
%                'parameters' Cell array of parameters to be tested, relative to Beta files.
%                            e.g. {[1 3],[2 4]} or {[1 3],[2 4];[1 3],[2 4]} in case of 2 groups.
%                            use ones for con files, e.g. {[1 1],[1 1]}
%                            Add nested cells for more repetition levels.
```

so for instance for a 2 x 2 x 2 you do 'parameters',{{[1 2],[3 4]},{[5 6],[7 8]}} basically the cell indicate the factors