sccn/eeglab #871: Mismatch between ERSP trial count from std_readersp and original EEG dataset trial count (EEGLAB)

TWXinPing, 2025-06-09:
Hi everyone,

I'm currently encountering an issue while using **std_readersp** to extract single-trial ERSP data for ICA cluster analysis.

I’m analyzing an EEGLAB STUDY with ICA clusters, and I extracted ERSP data like this:

> [STUDY, erspdata, times, freqs, erspbase] = std_readersp(STUDY, ALLEEG, ...
   'design', design_idx, ...
   'clusters', cluster_id, ...
   'component', comp_idx, ...
   'singletrials', 'on', ...
   'subbaseline', 'off');

The resulting ERSP has shape:

> size(erspdata) = [freqs × times × trials] = [83 × 200 × 175]

However, my original EEG dataset has:

> ALLEEG(31).trials = 178
STUDY.datasetinfo(31).trialinfo = 1×178 struct

So I’m missing 3 trials in the ERSP output.

What I’ve checked:
ALLEEG(31).reject is empty.

No error or warning was shown when running std_readersp.

The component used (comp_idx = 67) exists and works fine otherwise.

My question is:
Where could the missing trials have gone?
What internal mechanism in std_readersp or EEGLAB’s STUDY pipeline might cause this mismatch?

And how can I retrieve trial-by-trial ERSP data that fully matches my original behavioral trial info?

Additional Info:
EEGLAB version: 2023.0

Dataset is epoched and ICA-decomposed

All trials were labeled with behavioral info (custom trialinfo fields)

Any suggestions would be very helpful. Thank you in advance!

arnodelorme, 2025-06-12:
There should be no missing trials. 

[83 × 200 × 175]

175 is the number of subjects, unless you used the option "singletrials", "on" and have only one subject.

TWXinPing, 2025-06-13:
It's only for one subject. The variable  'component' assigns the component from specific subject within the cluster.

arnodelorme, 2025-06-13:
Can you share your data?

arnodelorme, 2025-09-10:
I have looked at the data in detail and could not find any issues. I tried plotting myself and got 178 trials.