sccn/eeglab #872: statcond - bootstrap - paired

arnodelorme, 2025-06-12:
I am a faithful user of EEGLAB and I am emailing you regarding the statcond function (bootstrap, paired options).

It is either that there is something wrong with this function (I hope not) or I have severe misunderstanding on what it does despite reading the function and its documentation. I spent four days examining this and I hope you can help me out.

I did a very simple test:

got two vectors of data:
x1=[1,2,3,4,5]
x2=[10,20,30,40,50]

and examined the variable res which is the output of the surrogdistrib function.
The options were: bootstrap, paird, and naccu=4(example)

This yielded a pool of data (4 x 5) with an unexpected pattern:
res{1} reshaped:
1 2 30 40 50
10 20 30 4 50
10 20 30 40 50
10 20 30 4 50

----
res{2} reshaped
1 2 3 4 5
1 20 3 40 5
1 2 3 4 50
1 20 3 40 50
---

arnodelorme, 2025-06-12:
This is what I got. How can I reproduce the problem?

```
x1=[1,2,3,4,5]
x2=[10,20,30,40,50]
[t df pvals surog] = statcond({ x1 x2 }, 'method', 'bootstrap', 'naccu', 4, 'paired', 'on')
1 x 2, paired data, computing T values
Using paired t-test
Bootstraps (of 4):...

t =

   -4.2426

df =

     4

pvals =

     0

surog =

  1×4 single row vector

    1.0000    1.0000   -1.0000    1.2910
```

arnodelorme, 2025-06-12:
Because the data is paired, the resampling is being done independently at each position, which differs from the classical paired bootstrap. Classical bootstrap resamples from the observed data and does not simulate the null hypothesis directly. By contrast, the permutation test directly generates the null distribution.

In EEGLAB, we use a modified bootstrap. The values are paired so in your case, you are doing 5 independent bootstraps. It is easier to reason with the first values in each group. So your data is 1 and 10. Paired. This is it. You cannot really do stat with 2 values, but if you could, your bootstrap null distribution would be a collection of 1 and 10. Bootstrap makes sense if you have 6 paired groups or more.

Now let's assume you have 10 groups (1, 10, 100, 1000, etc...), then bootstrap makes more sense. This is why when you use the GUI, you have a warning when selecting bootstrap. For 2 groups, permutation is better. 

What EEGLAB is doing (at least in the surrogdistrib function) is not a classical paired bootstrap. It performs independent element-wise resampling for each position.

Note that it would be quite different if the data were unpaired. For all practical applications, bootstrap for inference testing with paired data should be avoided in EEGLAB (there is a warning in the GUI that it is not recommended when you select it), and the default is to use permutation between conditions.

HaneenSu, 2025-06-13:
Dear Arno,

Thanks for this detailed explanation.

So if I want to recap what you said, within EEGLAB toolbox, the recommendation is below:
1) If i have paired data, then use permutation, always [regardless how many samples I have per group]
2) Bootstrap is suited when I have > 6 samples and **un-paired data.**

Last question, based on this, if I have two groups
x1=rand(30,1);
x2=rand(30,1);

and the values in x1 and x2 are actually paired.

Then, based on EEGLAB implementation that breaks the entity of pairing, using bootstrap for conducting a statistical test is methodologically wrong. Is this correct? If this statement is not entirely accurate, please let me know.

Thank you very much!!

Best
Haneen

arnodelorme, 2025-06-13:
I have looked a little bit more into it, and this type of bootstrap implemented a long time ago is not correct. I have removed it from the documentation. The correct way to do it would be to subtract x2 from x1 and bootstrap the values.